RobotLocomotion/drake #1749: Automatic lint on pull requests

david-german-tri, 2016-02-24:
Once we settle on a style in #1741, we should set up automatic linting on pull requests.  There are linters for both Google and ROS style.

jamiesnape, 2016-02-25:
I will add `cpplint` to Jenkins early next week.

jamiesnape, 2016-03-04:
There is now a nightly cpplint job on Jenkins. I will switch it on on pull requests early next week, once it is tested and a few minor issues resolved:
https://drake-jenkins.csail.mit.edu/job/linux-cpplint-nightly/

david-german-tri, 2016-03-06:
:+1:  (Also, wow, that's a lot of lint errors.)

jamiesnape, 2016-03-07:
It needs more work to filter out third party packages, but we also have include-what-you-use running:

https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-include-what-you-use/6/console

jamiesnape, 2016-03-08:
I think someone is going to have to fix some of the lint errors before we enable this on pull requests. At the very least, we need to resolve #1805.

jwnimmer-tri, 2016-03-24:
> ... fix some of the lint errors before we enable this on pull requests ...

Actually, I think a better way is to configure cpplint with enough suppressions so that it passes now, enable cpplint on PRs (to hold the ground we've already won), and then incrementally chip away at the suppressions until only the ones we are okay with remain.  I have a jwnimmer-tri/drake/tree/lint branch where I will start working on that now.

david-german-tri, 2016-03-24:
@jwnimmer-tri Seems like a great idea to me, as I sit here resolving a horrible merge because someone else didn't clang-format.

billhoffman, 2016-03-24:
Anther option would be to use pre-commit hooks in git.  For other projects at Kitware we have a Utilities/SetupForDevelopment.sh that you run from your git shell.  This puts checks in before people create the commits.

david-german-tri, 2016-03-24:
@billhoffman That would be a useful tool!  I'm not sure it's a substitute for PR-time checks, though, since we can't ensure all Drake contributors are using the pre-commit hooks (right?)

rpoyner-tri, 2016-03-24:
pre-commit hooks are a fine idea, but the best you could probably do is commit some preferred example hooks and try to persuade commiters/PR-ers to use them.

jwnimmer-tri, 2016-03-25:
Now that #1924 is in, possibly this can be enabled by default?

jamiesnape, 2016-03-25:
Sure. It is on my list of high priorities.

jamiesnape, 2016-03-25:
One warning to fix or suppress and then I can switch it on.
```
drake/systems/plants/test/testXmlUtil.cpp:16:  Missing space around colon in range-based for loop  [whitespace/forcolon] [2]
```
https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-cpplint-nightly-27&field2=buildstarttime&compare2=84&value2=now

rpoyner-tri, 2016-03-25:
PR at #1946 .

jamiesnape, 2016-03-28:
Lint is enabled for all new PR builds: https://drake-jenkins.csail.mit.edu/job/experimental/compiler=cpplint,label=linux/

jwnimmer-tri, 2016-04-07:
This seems to be working well.  Is there any more to be done?

jamiesnape, 2016-04-07:
I do not think so.

jwnimmer-tri, 2016-04-07:
Over to @david-german-tri to assess the outcome, and close, then.

david-german-tri, 2016-04-07:
Fantastic.  This thread got lost in the shuffle and I never noticed the new line in the Jenkins matrix.   Thanks, @jamiesnape!

closed 2016-04-07 by david-german-tri