RobotLocomotion/drake #1759: ros messages for csail-toyota simulator

RussTedrake, 2016-02-26:
from felix:

it will probably be helpful to have these msgs types available in the simulator:
http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html
http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html
http://docs.ros.org/api/geometry_msgs/html/msg/PoseWithCovarianceStamped.html
http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html (edited)

liangfok, 2016-09-08:
So far, odometer, laserscan, and PoseWithCovarianceStamped message are being published. Versions of the publishers that support a `RigidBodySystem` with only a single model instance are in master:

* [odometer](https://github.com/RobotLocomotion/drake/blob/master/ros/drake_ros_systems/include/drake/ros/systems/ros_sensor_publisher_odometry.h)
* [laserscan](https://github.com/RobotLocomotion/drake/blob/master/ros/drake_ros_systems/include/drake/ros/systems/ros_sensor_publisher_lidar.h)
* [pose](https://github.com/RobotLocomotion/drake/blob/master/ros/drake_ros_systems/include/drake/ros/systems/ros_sensor_publisher_odometry.h) 

Versions that support multiple model instances are in my feature branch [feature/multi_car_sim_2](https://github.com/liangfok/drake/tree/feature/multi_car_sim_2). I originally attempted to merge them in PR #3103 but decided to split the PR into smaller ones, which is still a work in progress.

I have not begun work on the IMU messages.

liangfok, 2016-09-08:
Note that the odometer messages currently do not include covariance matrices: https://github.com/RobotLocomotion/drake/issues/2457.

jwnimmer-tri, 2017-01-13:
@liangfok Does this meta-issue still add value?  I'm not sure its really up to date, and I'm not sure that fixing the gaps is on the roadmap.  So at least demote to `priority: backlog`?

liangfok, 2017-01-13:
This is another issue awaiting porting to System 2.0. I'll keep it at medium priority since people are asking for it.

jwnimmer-tri, 2017-02-09:
@liangfok I think that priority should represent the priority that we will treat it with, not the priority that it is being requested with.  As such, I think "backlog" more accurately expresses that we are not going to work on it soon.

closed 2018-06-09 by jwnimmer-tri