RobotLocomotion/drake #1763: design support for autodiff relative to parameters in RigidBodyTree

RussTedrake, 2016-02-27:
There are a number of use cases where we would like to get the gradient with respect to some parameters -- two that come up in the matlab code are system ID and kinematic design optimization.  We need to think through how we are going to implement this in c++.

It's relatively simple for me to imagine templating the RigidBody class on ScalarType and having the inertial parameters potentially become autodiff.  Then we would want to design the interface -- off the top of my head i could imagine:
  - in URDF/SDF, look for a "parameter" attribute on any given field to label that as a parameter of interest (we will not want autodiff for every element of the spatial inertia).  
  - design a codepath with support methods to parse the urdf/sdf and instantiate with autodiff RigidBody elements, properly initialize the autodiff variables given the subset of parameters, run some code, then extract the gradients.

The same thing should probably work for the kinematic parameters, but it would have to get piped through the KinematicsCache object, too (which is already templated, but the transforms it builds on are not).  

Perhaps an even simpler place to get started, and one that might have a lot of practical value, would be to work through a case were the only parameter is, say, the damping in a joint (also easy to template).  

@tkoolen -- would value your thoughts here.

tkoolen, 2016-02-27:
> templating the RigidBody class on ScalarType
Yep, I think that's unavoidable. It would mean templating DrakeJoint and RigidBodyTree on the scalar type as well.

> in URDF/SDF, look for a "parameter" attribute on any given field to label that as a parameter of interest (we will not want autodiff for every element of the spatial inertia).

Probably what you were thinking too, but to clarify then: I think we should create an autodiff rigid body tree with all rigid bodies and joints instantiated as autodiff bodies and joints and with all their properties instantiated as autodiff types if gradients w.r.t. any of the model parameters are required. The parameter tags in the urdf should simply determine the meaning of the derivatives vector. So if there's only one parameter in the URDF, the derivative vector for all of the properties of all joints and bodies will have length one, and will correspond to variations in that one parameter.

Something we will need to change is that we're currently assuming that the scalar type of the KinematicsCache object dictates what scalar type should be used for all computations (e.g. in the center of mass method). That can no longer be the case, and we'll have to use a template trick to figure out what the appropriate type used for calculations should be.

RussTedrake, 2016-02-27:
Exactly right.

aespielberg, 2016-02-29:
I'm jumping in here and admittedly don't have a great grasp on how the autodiff architecture works yet, but thought I'd chime in since this is exactly the functionality I need for scaling up the aforementioned robot design optimization stuff.  Correct me if I say something stupid :-).  I see this is exploiting functionality built directly into Eigen, and it looks like it's all numerical.  This is probably very good, since I've found symbolic dynamics to be MB long files when not simplified (a nontrivial operation), so a symbolic solution probably won't work here.

Twan's architecture is exactly the one that would be needed for something in design optimization.

I assume that the way this is structured, the output of functions such as .dynamics and forward kinematic routines would be autodiff as well?  This would allow users to easily build objective functions which are autodiff as well.

Incidentally, autodiff on inertia tensor would be very useful (which it's parameter-based).  Is there a reason you want to avoid this?  For instance, there are cases where one would want to change the dimensions of a cylindrical link; in order to keep the inertia consistent with the changing geometry, it should be a function of a (say) mass parameter and geometric parameters.

I agree (if I'm understanding Russ's suggestion correctly) that it's important to differentiate between variable parameters and fixed parameters.

RussTedrake, 2016-02-29:
Thanks Andy. We absolutely do intend to include inertial parameters. I think
you’re referring to my comment, but I just meant that to say that we might not
want to parameterize every element of the spatial inertia matrix, but often just
a few components.

aespielberg, 2016-02-29:
Oh, then I definitely agree.  The use case where everything is parameterized is either typically pretty esoteric, or a very interesting/difficult research project.

aespielberg, 2016-03-20:
I am wondering if this can be assigned to me, especially since Twan is busy implementing other stuff right now.  This is something I need, preferably soon, for my research, and since I'm publishing in this space I think it would make sense for me to work to implement it, if that is okay with people.  I'm new to the C++ drake dev, but I figure this would probably be a good experience to learn the internals well.

RussTedrake, 2016-05-08:
tons of progress on this in #2267

RussTedrake, 2016-11-27:
@amcastro-tri has made a first PR on this.  will assign this to him to close when appropriate.

amcastro-tri, 2016-11-28:
Thank you @RussTedrake. #3987 is tracking the progress of this work which is exactly the same as proposed at the very beginning of this thread, templating RBT on `<T>`. 
I believe then this is a duplicate of #3987 and since #3987 provides a detailed tracking of progress I will close this issue.

closed 2016-11-28 by amcastro-tri