RobotLocomotion/drake #1766: nightly builds / releases

RussTedrake, 2016-02-27:
Moving the last remaining discussion from #1526 (and closing that issue).

everyone agrees we want nightly releases, and we need build servers running tests on those releases (ideally before it was posted, but github has a webhook for release postings that could make it easy to do things serially).
Specifically:

 - [] nightly build should replace a single "nightly" release on the github release page.
 - [] we can tag a good release (and increment the version num)
 - [] we carry a branch "stable" which gets moved up to the latest release, and the default install instructions tell people to clone stable.

jamiesnape, 2016-03-01:
Jenkins can post nightly builds on success. Ideally, I would post these to an S3 bucket with a public link. You could place a link the latest nightly on the wiki. I would reserve the GitHub releases for the stable versions.

RussTedrake, 2016-03-02:
Great. Let me know the link.

[https://github.com/RobotLocomotion/drake/issues/1766#issuecomment-190779847] .[https://github.com/notifications/beacon/AGJNNPeC8H7x8xD4xib87yYwCl3vY7PUks5ppGClgaJpZM4Hkeks.gif]

jamiesnape, 2016-03-02:
Will do. To get this fully functional, we first need to resolve #1751 and #1781.

billhoffman, 2016-03-02:
Work on a partial one for linux that uses the shell scripts.

RussTedrake, 2016-03-16:
note: have to be very careful to not include proprietary source.  (e.g. snopt).  we have some “release_filelist” logic in the make files which help with this.  i can also share our existing packaging scripts.

jamiesnape, 2016-03-16:
Are they the scripts in the admin repo?

RussTedrake, 2016-03-16:
Yes.  In the www subdir

RussTedrake, 2016-03-16:
Make that packaging subdir :-/

RussTedrake, 2016-05-01:
the other thing that we will want to be conscious of (eventually) is to have the minimum required versions for all of the libraries on the servers that build the official releases.  (see e.g. #1428)

david-german-tri, 2016-06-01:
Downgraded to medium because this is blocked on having code that's stable enough to release.

jwnimmer-tri, 2016-11-08:
So here's a good question.  Should the releases be the C++ and Python code only, or also include the MATLAB support?  It seems like a big delta on effort to add MATLAB.

RussTedrake, 2016-11-08:
I’ve been leaning to not including the matlab, just to get these up.  We’ll still need to supply some files from the source directory (urdf + meshes), but the list should be pretty reasonable.  Will also have to address path relocation issues.

jamiesnape, 2016-11-08:
I will start with C++ and take it from there. Dealing with the pure CMake projects should be relatively easy, since the functionality for creating packages is built-in. It is mostly a case of ensuring that everything that needs to be in the installer has an `install` command in front of it.  If projects (especially Drake) are not using CMake life gets more difficult.

jamiesnape, 2016-11-08:
The other question is which externals would you like to include. Given the variety of licenses, that impacts packaging and distribution.

RussTedrake, 2016-11-10:
Looks like it might be worth cleaning up a little more on #4045, like setting the eigen flag and resolving the thirdParty issues in drake.  (plus resolving, or explaining, the remaining ???s)

It looks like we can get pretty far using drake + the BSD-3 and Zlib (bullet is worth including) licensed externals.  How restrictive is Zlib?

jwnimmer-tri, 2016-11-10:
IMO Zlib is comparable with BSD-3 -- you have to be careful about labeling and notices, but it's permissive use otherwise.

jwnimmer-tri, 2016-11-10:
Also FYI, the Eigen MPL-only flag is done (per #4065).

RussTedrake, 2016-11-10:
Great.  Can we update #4045 with both of these comments?  (sorry I asked on this thread instead)

jamiesnape, 2016-11-10:
I added a few notes. Bullet should be fine, I just need to double check we do not use the small amount of code under different licenses.

jamiesnape, 2016-11-10:
If we need a solver, including part of ipopt is not out of the question. It just depends on the minimum functionality that is needed.

RussTedrake, 2016-11-10:
sorry, i forgot.  We have permission fromthe author to redistribute code linked (statically) against snopt, but not to distribute the snopt libraries themselves.  We should definitely handle that case here (snopt is the solver of choice)

jamiesnape, 2016-11-10:
Cool. That makes life easier.

jamiesnape, 2016-11-10:
(I am assuming no SNOPT headers may be distributed, which is not an issue at present.)

jamiesnape, 2016-11-10:
@RussTedrake Would you like to distribute binaries of the examples, or just libraries?

RussTedrake, 2016-11-10:
definitely examples, too.  thanks!

david-german-tri, 2017-01-13:
This is actively being worked, bumping the priority.

david-german-tri, 2017-01-26:
Note from f2f: We will need to make sure that build tree RPATHs do not leak into the installed package.

patmarion, 2017-02-23:
btw, regarding RPATH and redistributable packages, Director has packaging for linux and mac that fix up RPATH and I am quite happy with how it works!

https://github.com/RobotLocomotion/director/tree/master/distro/package

jamiesnape, 2017-02-23:
Cool. Looks useful.

patmarion, 2017-03-02:
fyi, I just tested Director's packaging script on the drake superbuild.  It successfully makes redistributable packages for director that also include all of drake.  If you want nightly packages/releases, I'd recommend turning this on as a starting point.

stonier, 2017-07-31:
Jenkins now has a set of jobs under [Packaging](https://drake-jenkins.csail.mit.edu/view/Packaging/).

EricCousineau-TRI, 2018-02-02:
BTW Should we consider this closed for now?
While explicit tagging is not being done, we can just "manually" tag the releases, and possibly mirror the release to a different URL if so desired?

jamiesnape, 2018-02-02:
This issue about nightly releases, so tags and the whole stable release process would be different issue anyway.

EricCousineau-TRI, 2018-02-02:
Sounds good! Closing for now.

closed 2018-02-02 by EricCousineau-TRI