RobotLocomotion/drake #1767: drake-cdash-bot is loud

tkoolen, 2016-02-27:
I'm a strong proponent of restricting the conversation section of pull requests to human authors only. The standard github API for build server notifications is a much better solution than automatically posting comments to the pull request in my opinion.

jamiesnape, 2016-03-01:
You will lose the direct links to CDash and will have to click through at least three pages to find the build error.

jamiesnape, 2016-03-01:
Note, also the drake-jenkins-bot has to be able to comment, so we cannot restrict to human authors.

jamiesnape, 2016-03-01:
@RussTedrake I have disabled the bot. Let me know if you change your mind.

billhoffman, 2016-03-01:
Maybe put them all under one comment.

billhoffman, 2016-03-02:
Look at making it less verbose.

david-german-tri, 2016-04-25:
No one except me seems to miss this, and now that I've learned my way around the Jenkins UI I don't even miss it very much.  Resolving.

closed 2016-04-25 by david-german-tri