RobotLocomotion/drake #1769: Define which libraries and headers constitute the Drake API

david-german-tri, 2016-02-27:
What is the stable interface the Drake library presents to users?  (library targets, .h files, function signatures)

Motivating example: Right now, the drakeLCMSystem library is not exported in CMake, so an outside dependency that wishes to use drakeLCMSystem must enumerate all of the transitive dependencies by hand in order to link.  If drakeLCMSystem is intended to be private, that's fine, but if drakeLCMSystem is intended to be part of the API, it should be exported.

See also #1762.

RussTedrake, 2016-02-27:
This is one of the biggest and most important things that we need to clean 
up in
the redesign (as discussed in #1525). It was not something we could 
architect in
Matlab, so never got architected.

jamiesnape, 2016-03-07:
This is really necessary for #1751 and #1781.

iamwolf, 2016-03-07:
(drakeLCMSystem is now exposed as part of drakeRBSystem, cf. https://github.com/RobotLocomotion/drake/pull/1754/files)

RussTedrake, 2016-05-08:
marked as high priority because it's super important, but i think it makes sense to wait until system 2.0 lands (and perhaps some other big changes, as well) before we move on this.

david-german-tri, 2017-04-13:
@stonier I think this issue is finally starting to come into focus.  The short-term plan for `package_drake.sh` is to include everything and the kitchen sink, but at some point someone needs to curate the APIs that we package into an actual product.

jwnimmer-tri, 2018-06-09:
I think issue is probably finished, and "the things installed by Drake's binary release" is the short answer we've settled on, but I will reconsider in the next couple weeks if there is more still to be done.

jwnimmer-tri, 2018-06-14:
I don't there is more to do right now, but @RussTedrake please re-open if you disagree.

closed 2018-06-14 by jwnimmer-tri