RobotLocomotion/drake #1772: wrap LCP solvers c++ optimization hierarchy

RussTedrake, 2016-02-28:
- Evan Drumright has code he is willing to share
- siconos has a nice set of solvers

needed by #1594

RussTedrake, 2016-03-05:
Here is the code from @edrumwri 
https://github.com/PositronicsLab/Moby/blob/master/src/LCP.cpp

Evan is excited to collaborate with us on this, and we're thinking the best way to start is port and adapt this implementation into our own LCP.cpp in drake/solvers/ and connect it up as a solver that is dispatched to from the OptimizationProblem class.  To make sure Evan get's credit, we will put a few lines on the top saying something like "Modified with permission from code by Evan Drumwright (https://github.com/edrumwri)".

RussTedrake, 2016-03-05:
Specifically, I'm imagining a new method,
```
Constraint& addLinearComplementarityConstraint(Matrix<> M, Vector<> q, DecisionVariableList z)
```
which matches the interface here: https://github.com/RobotLocomotion/drake/blob/master/drake/thirdParty/path/pathlcp.m .  (note the above is just intended to be pseudocode, actually want const&, etc).  

and new MathematicalProblem class for LinearComplementarityProblem.  The calling solve on a LinearComplementarityProblem should dispatch to one of these two solvers.

RussTedrake, 2016-03-05:
Section 10.2 of http://link.springer.com/book/10.1007%2F978-1-4757-3040-1 has some good candidates for examples/unit tests.  (@Lucy-tri has the pdf)

RussTedrake, 2016-03-06:
i was just given a new zip with win64 libs for PATH 4.7 by Steven Dirske.  if we can get the other platforms, too, then that might be viable, too.

psiorx, 2016-03-06:
> i was just given a new zip with win64 libs for PATH 4.7 by Steven Dirske

Nice.  The unavailability of 64-bit libs is what was holding us back before.  

> siconos has a nice set of solvers

For the record, the issue that was preventing us from using siconos in place of path was that their LCP and MLCP solvers don't support negative lower bounds on the decision variables which our formulation currently requires.

RussTedrake, 2016-03-06:
I think “requires” is not quite right. We took advantage of the MCP, but can
easily re-formulate around that.
I have the windows, linux, and mac (64-bit only) binaries for PATH 4.7 now, too.

edrumwri, 2016-03-07:
I'll give a little more info on why Moby uses pivoting solvers, rather than providing Siconos' arsenal of solvers. For the past 3-4 years, I've focused on accuracy and have targeted small applications (like a robot grasping or walking) rather than massive numbers of interacting bodies (Anitescu does the best work there).

One initial note: the matrices in the LCP/NCP problems can be very sparse, depending on the number of bodies under consideration: if there exist contact constraints between bodies 1 and 2 and between bodies 2 and 3 but not between bodies 1 and 3, the entries in the matrix corresponding to pair 1/3 will be zero. 

The siconos solvers run the gamut from pivoting ("direct" to use an analogy to numerical linear algebra), to matrix splitting methods ("iterative" by the same analogy), to Newton-based (either optimization or nonlinear system solving). 

pivoting: enumerative
matrix splitting: pgs, rpgs, psor, block
unknown: qp
Newton: PATH
other: nsqp, cpg (seems to exhibit the worst traits of pivoting and iterative)

The matrix splitting method solvers do not converge particularly well (Lacoursiere has a nice paper on this, see below). For time stepping simulations, the users are often directed to tune parameters for constraint stabilization in order to combat the residual error from lack of convergence. Too little gain yields significant interpenetration, while too much gain results in bodies popping apart.  

Pivoting methods and active set QP solvers yield very low constraint error, but they are limited in simulation to ~1500 contacts (assuming only a friction pyramid). One of my solvers is able to exploit an upper bound on the number of points of contact between two bodies, but many bodies contacting simultaneously will still cause running times to go through the roof: sparsity is not exploited. I've looked into incorporating sparse direct linear algebra solvers into the pivoting algorithms, but the libraries for the former are not yet mature.

I've had some success with Newton-based methods (interior-point solvers, in particular), as described in my paper (reference below). Sparsity can be exploited. If the contact problem is "relaxed" to a convex QP, the resulting constraint error is larger than under pivoting methods but smaller than under matrix splitting methods. PATH seems to be a very robust solver (it should be, as it is meant to be commercial grade), and compares well to Lemke's Algorithm (see my other paper below). Its NCP solver can reduce the number of mathematical programming variables significantly if a true friction cone is desired (there are no problems that I'm aware of that exhibit divergent artifacts without a true friction cone).  One philosophical concern: PATH cannot guarantee solution to the NCP/LCP problems (though my paper indicates it works very well!)

TL;DR: I recommend pivoting methods for few bodies in contact in an island and Newton-based approaches for many bodies in contact simultaneously. I recommend matrix splitting methods only for applications where speed is much more important than accuracy and extensive tuning can be performed.

References:

[Lac03] Claude Lacoursi`ere. Splitting methods for dry frictional contact problems in rigid multibody systems: Preliminary performance results. In M. Ollila, editor, Proc. of SIGRAD, pages 11–16, Nov 2003.

[DS10] Evan Drumwright and Dylan A. Shell. Modeling contact friction and joint friction in dynamic robotic simulation using the principle of maximum dissipation. In Proc. of Workshop on the Algorithmic Foundations of Robotics (WAFR), 2010.

[DS12] Evan Drumwright and Dylan Shell. Extensive analysis of linear complementarity problem (LCP) solver performance on randomly generated rigid body contact problems. In Proc. IEEE/RSJ Intl. Conf. Intelligent Robots and Systems (IROS), Vilamoura, Algarve, Oct 2012.

RussTedrake, 2016-03-07:
Very nice.  Thanks @edrumwri

vacary, 2016-03-09:
@psiorx Could you give more details about the fact that siconos does not support negative lower bounds in the LCP and MLCP solvers ? Do you want to solve Affine Variational Inequality (AVI) on a box, or just to be able to put small lower bounds in the resolution of your LCP. In the latter case, why not shifting your data ?

We are really interested in getting feedback on Siconos. Do not hesitate to send your issues or comments on siconos ?

RussTedrake, 2016-03-09:
@vacary -- great to see your comments here! 

our time-stepping implementation also handles some equality constraints and joint constraints which are currently implemented as mlcp constraints with both upper and lower bounds.  We can definitely implement them as lcp constraints by shifting the data and potentially doubling the number of constraints.  this is why I said above that a simple reformulation can put it back into lcp.  in fact, the mlcp interface we had been using in path might have been doing this behind the scenes.  I don't think there are any fundamental blocks to trying the siconos solvers.

vacary, 2016-03-09:
@RussTedrake It is my pleasure :-)

We have the same type of constraints in [siconos](http://https://github.com/siconos/siconos) when we want to deal with friction with constant threshold and also stops in joints. We do the transformation most of the time by hands.

Nevertheless, if your problem is of the form 
-(Mz+q) \in N_[l,u](z)
with l,u some vectors in RR^n of lower and upper bounds, we use the Relay problem in Siconos for solving it. I think it could be of interest to treat explicitly the case of equality constraints u_i=l_i and the case of standard complementarity u_i =+ \infty or l_i =-\infty for some i.

I will discuss tomorrow with Olivier Huber @xhub about a more general answer to the questions and comments of @edrumwri . In particular, we have also a general mechanical problem that enables to mix several type of constraints, but it can be improved in several aspects.

edrumwri, 2016-03-10:
Cool! I would very much like to hear @vacary<https://github.com/vacary>'s analysis of my analysis!

vacary, 2016-03-11:
A first comment on friction pyramid and second order cone. The main drawback of friction pyramid, or faceted cone, is the introduction of anisotropy in the friction response. The fact that the frictional force is not collinear to the sliding velocity is for me really a huge problem. It is direct consequence of the maximum dissipation principle over a polyhedron. When we slide, we slide on a surface with a network of oriented grooves. This price to pay to get a proof of convergence of LCP solvers is for me too high on the mechanical point of view.

In siconos, we decided to separate clearly LCP, MLCP and Relay solvers from the friction contact solvers (fc2d and fc3d) on the second order cone .

 As mentioned @edrumwri , for LCP we have the following list of solvers
- enumerative, branch and bounds (CPLEX)
- std pivoting Lemke and murty and PATH
- splitting/projection solvers: PGS, ...
- solvers inherited from VI formulation : FP, Extragradient as so on
- Newton--based solvers : Min, Fisher burmeister, 

We never complete a efficient implementation of an interior point solvers. Perhaps, it could be added on the TODO list.

 For fc3d, we have the following list of solvers
- projection/splitting (nsgs) with a bunch a local solvers : Newton, analytic, projection, extra gradient
- Newton based solvers with various formulations.
- proximal point algorithm
- solvers inherited from VI formulation : FP, Extragradient as so on

Our experience is as follows:
- iterative solvers based on projection/splitting or VI solvers are the  more robust solvers, even if the convergence is slow (high accuracy is very expensive). Bullet and Chrono::Engine use such type of approaches. We tend to use it even for small problems that comes from robotics where a large number of iterations is not so expensive. 
- Newton-based methods converges super linearly or fails very quickly. In the case of rigid body mechanics, if there is some hyperstaticity, Newton solvers often fails. If you have more contact points variables that degree of freedom, Newton based solvers will fail. The same conclusion has been made by Kleinert et al. for Interior point solvers.

vacary, 2016-03-11:
Concerning the solvers in Chrono::Engine and the fact that Anistescu and Tasora do the best work for large collections of bodies. It is clear that Chrono is really very good to take benefit from parallelism and GPU computing. But you have to be aware that they solve the convexified contact problem with associated friction. The consequence is that you have some dilatation in the normal direction when you slide. For grasping or walking, it is not acceptable. 

@edrumwri : I am not sure to understand the link between the solver accuracy of the friction contact problem and the stabilization of constraints when you cite Lacoursiere. Usually, when we deal rigid body with contact, impact and friction, time--steppers solve the problem at the velocity level. Then you have some drift into the constraints which is mainly due to the time--integration approximation rather the accuracy of the friction contact solvers. In a second step, you have to treat the violations by projection or baumgarte stabilization. A accurate solution of the friction contact problem will not improve the violation of the constraints. Do yo solve the problem at the position level ?

sherm1, 2016-03-11:
> A first comment on friction pyramid and second order cone. The main drawback of friction pyramid, or faceted cone, is the introduction of anisotropy in the friction response. ... When we slide, we slide on a surface with a network of oriented grooves. This price to pay to get a proof of convergence of LCP solvers is for me too high on the mechanical point of view.

Yay -- music to my ears! I agree wholeheartedly.

sherm1, 2016-03-11:
> We are really interested in getting feedback on Siconos. Do not hesitate to send your issues or comments on siconos.

Siconos appears to be GPL licensed -- if so we can't include a line of it in Drake :cry:.

RussTedrake, 2016-03-12:
We may be able to link to it as an optional external, with the understanding that if someone selects that option it would instantly infect the rest of the codebase.  But it would be very nice to compare solvers properly.

xhub, 2016-03-12:
We are in the process of changing our license to LGPL, see https://github.com/siconos/siconos/issues/7

Then would that be good enough for using it from Drake?

xhub, 2016-03-12:
Just to complement @vacary 's answer to @edrumwri

I'm not sure we have the same terminology, but Lemke's algorithm for me is a pivoting method, for which it is possible to exploit sparsity and do rank-one updates (cf LUSOL, the linear algebra engine of MINOS). Saunders also published LUMOD, as set of routines to do Block-LU updates. So sparsity can be efficiently exploited (PATH always works with sparse matrices) with any linear
algebra package (UMFPACK, MUMPS, PARDISO).

Also PATH is based on Lemke's algorithm. It has a lot a fancy things build on top (crashing method to find a good starting point, proximal perturbation) but via it's option mechanism you can make it use only its pivoting method. Based on my small testing, the crash method is not always a good option and can be quite slow. But on some instances, it can alone solve some problem. So it's hard to draw conclusion, even if in the case you want really accurate solutions, I found that it is better to use the ``pure'' Lemke algorithm.

The nonlinear version (NCP) is based on Newton approximation in a very specific sense, that is it solves a sequence of LCP. Then there are again some strategies to save the day when the problem is ill-conditioned or the theory may not apply. Still for both LCP and NCP, the algorithm used by PATH has a strong convergence theory[1] [2], even though it is not always trivial to check the conditions ...

[1] http://www.cs.wisc.edu/~ferris/papers/mor-avi.pdf
[2] http://dx.doi.org/10.1287/moor.19.2.352

edrumwri, 2016-03-12:
Thanks for the feedback, @vacary. Glad to hear that I didn't characterize anything incorrectly- differences seem to be mostly philosophical, and I'll pontificate on some of these below.

One thing that I hope is apparent to readers is that there is no free lunch, and significant research questions remain. Building solid multi-body simulations with contact is not just a question of solid software development practices (which is an assumption that many outside the research community make).

Our paper [DS12] (cited earlier in the thread) lends evidence to @vacary's observation that interior point methods are sensitive to "redundant" points of contact, but our experiments (experimental data is available and new data can be produced using source code from our [project page](http://robotics.gwu.edu/positronics/wp-content/uploads/2013/12/LCP-eval/) indicate that PATH performs very well on such problems.

On to philosophy:

I like the pivoting-based approaches because I prefer the projected constraint velocities to be solved as accurately as possible. I acknowledge that approximating the friction cone comes with an accuracy cost, though my point was that "there are no problems that I'm aware of that exhibit _divergent_ artifacts without a true friction cone". I use "divergent" because the context is robotics, so controlled systems with nonsmooth behavior might lead to control challenges. That then leads to the question: "what about uncontrolled systems"? For a truly accurate solution, it is clear that an elliptical friction cone is necessary. I'm not ruling out that an elliptical friction cone is required to yield phenomena observed in real systems, but friction is a complex, microscopic phenomenon and Coulomb friction is one of the simpler friction models (albeit still challenging to work with). To add one more complication, time stepping approaches must select a tangent direction to apply the frictional forces: should the force oppose the velocity at time t or the velocity at time t+\Delta t (I know this will add to @sherm1's skepticism of time stepping!)? 

With regard to @vacary's question about solver accuracy vs. constraint stabilization: I am not considering positional approaches. The statement "you have some drift into the constraints which is mainly due to the time--integration approximation rather the accuracy of the friction contact solvers", depends on the relative sizes of the geometries and the size of the time steps. This statement is true for many problems in which the human eye is able to identify artifacts from animations. I have been investigating problems for which large differences in scale can generate significant artifacts, however.

edrumwri, 2016-03-12:
Ah, just saw @xhub's feedback. Some thoughts:

I investigated pivoting approaches with sparse linear algebra about three years ago, but every library I examined (UMFPACK, SuperLU, several others) presented several challenges that led me to abandon the effort. Most of the libraries I examined were not robust (compared to LAPACK), but maybe it's time to look into this again.

I did not know or had forgotten that PATH is using a different algorithm for LCP vs. NCP- good to know. One important note: globally convergent algorithms are capable of converging to a local minimum (despite the seeming contradiction), meaning that the NCP may not be solved by these algorithms. For building robust software, I like LCPs, because greater theory has been established at this point.

sherm1, 2016-03-12:
> @xhub: We are in the process of changing [siconos] license to LGPL. Then would that be good enough for using it from Drake?

That would be an improvement but it still imposes obligations on the behavior of downstream users of the code, which can be very difficult to ensure in a large organization. For example: strict separation between the LGPL code and the rest of the code must be maintained forever, and a mechanism must be in place to publish modifications in case a need to modify the LGPL code arises. That creates liability issues and a need for bureaucratic procedures that ultimately reduce the use of the code. We would like people to eagerly use Drake with no reservations.

My question would be: why do you want to control the behavior of other people? That goal competes directly with the goal of maximizing the use of your software, which is a common desire for many programmers. An alternative would be to release the software under the Apache 2.0 like the Apache Foundation and Google. That maximizes the software uptake, but in exchange you give up control. I think it's good tradeoff!

sherm1, 2016-03-12:
> @edrumwri: time stepping approaches must select a tangent direction to apply the frictional forces: should the force oppose the velocity at time t or the velocity at time t+\Delta t (I know this will add to @sherm1's skepticism of time stepping!)?

Not really! I think it's the wrong question to ask though -- it would be better to ask about stability vs. cost. The principal of maximum dissipation requires that friction forces *always* oppose the velocity. Those forces produce accelerations that appear in the differential equations that govern system motion. Time stepping is a way of solving those equations approximately by discretizing, and is expected to yield a result that is first-order accurate in step size dt. Different first-order discretizations exist that have different stability properties. Choosing the end-of step velocity would improve stability, but at the cost of making the step solution implicit. In fact if all the velocity equations are solved at t+dt (like implicit Euler) then the overall method becomes symplectic Euler and would be very nicely behaved! (Except for the nonlinear equation solve required at each step :cry:)

Note that if you choose v(t) when v(t)=0 friction is ignored entirely for that step. That could result in a large, qualitatively wrong error. That *probably* won't matter much for ongoing contacts but I think it is unacceptably awful for impacts -- see endless nagging about that in [Uchida et al. 2015](http://rspa.royalsocietypublishing.org/content/471/2177/20140859). 

As a further plug let me add that taking error controlled steps is an easy cure for this problem. The error estimator would measure a large error in a step where v(t)==0 and v(t+dt)>>0 and choose a smaller dt across the transition from sticking to sliding.

edrumwri, 2016-03-13:
I think communication breakdown somewhere- the velocity equations are solved at t+dt via the LCP/NCP. I was saying that if one looks at the time stepping equations, the frictional forces can be made to oppose the velocity at v(t) or v(t+\Delta t) or, even 0.5(v(t)+v(t+\Delta t)). I presume that the problem goes away as \Delta t \to 0, but I would need to go read the time stepping convergence proofs to verify.

Also, for v(t)=0 (I'm also assuming that you mean that tangent velocity is zero) for impacts, you are assuming a incremental collision model? For an algebraic one, the velocity changes instantaneously, so I don't see how the problem would apply.

I don't think you want to jump across discontinuities and use error control to manage the issue. Treating the problem as piecewise DAE's requires stepping to the event, as far as I know. Convenient that @vacary is on this thread- he's done some work in this area- so perhaps he wants to comment further.

vacary, 2016-03-14:
@sherm1 
> @sherm1: An alternative would be to release the software under the Apache 2.0 license like the Apache Foundation and Google. That maximizes the software uptake, but in exchange you give up control. I think it's good tradeoff!

I am pretty convinced by your argumentation. The debate is launched inside the contributors of siconos. I will let know our final choice. Hope it will Apache 2.0 ("The license").

sherm1, 2016-03-14:
Thanks for considering it @vacary -- that would be great!

ggould-tri, 2016-03-16:
A straightforward port of @edrumwri's LCP solver from Moby to the Drake optimizer stack is available in PR #1874

RussTedrake, 2016-04-30:
the immediate need is resolved.  will likely bring up the richer discussion about different solvers/formulations again, but i believe we can safely close this issue.

closed 2016-04-30 by RussTedrake