RobotLocomotion/drake #1774: missing java on first compile gets into trouble

RussTedrake, 2016-02-29:
A few users have compiled w/o java initially.  They get all the way through to the drake subproject, then get a cmake error because java is not found.  So they install java, but then get an error because libbot was configured without java, but drake finds java, so expects libbot's java support.

shouldn't be too hard to clean this up.  probably best if we check for java at the drake-distro level and error out immediately (before they get to compiling libbot).

jonathan-owens, 2016-03-01:
Easy enough to add -- before I commit and create the pull request, do we want to enforce a minimum Java version?

billhoffman, 2016-03-01:
@jonathan-owens I would not make  any  changes to versions.  Currently it is just a find_package(Java REQUIRED) at the drake level.  If you just move this up to the top superbuild level it should solve the problem.

jonathan-owens, 2016-03-08:
I think I fixed this -- works on my Ubuntu box, but I want to test it. Do I have to do a pull request to trigger the test servers? I thought we could do test this please or something of the sort...

jonathan-owens, 2016-03-23:
Closing this issue since #1871 was merged.

closed 2016-03-23 by jonathan-owens