RobotLocomotion/drake #1782: install Matlab on AWS instances for Jenkins testing

billhoffman, 2016-03-01:


billhoffman, 2016-03-01:
@RussTedrake and @jamiesnape, I tried to use the license.lic file and key.   However, sadly it does not work.  After typing in the license key and giving it the path to the license.lic file it complains that the license.lic was generated for a different computer:

[image]

Might be that we have to run a license server to get this to work.

RussTedrake, 2016-03-23:
we got the licenses for this, and @jamiesnape has made progress.  reassigning this to him.  @jamiesnape would love an update on this.

jamiesnape, 2016-03-23:
We have nightly and continuous MATLAB builds for Linux and GCC running on Jenkins now. Some more externals need switching on and I need to finish porting the known issues script. After that we will be good from a technical point of view.

jamiesnape, 2016-03-23:
Licenses expire in 18 days.

RussTedrake, 2016-03-23:
great. thank you

billhoffman, 2016-04-12:
Waiting on Matlab working on windows in CI.  Which is waiting on not using perl.

jamiesnape, 2016-04-22:
MATLAB is installed. See #2155 and #2156 for follow up.

closed 2016-04-22 by jamiesnape