RobotLocomotion/drake #1784: Drake should support submeshes in SDF

david-german-tri, 2016-03-01:
Right now, every time a .sdf file cites a [named submesh](http://www.sdformat.org/spec?ver=1.6&elem=geometry), Drake opens and parses the mesh URI, and includes the entire mesh in the element geometry.

This is both incorrect because you wind up with lots of unintended objects in the RigidBodyTree, and inefficient because the entire mesh is ingested and stored redundantly.

patmarion, 2016-03-07:
Is submesh a concept defined by collada files?  Is submesh supported by any other file format?  In other words, does this feature require .dae files and is incompatible with other mesh file formats?

david-german-tri, 2016-03-07:
obj files have groups.  In any case, I'd propose that the SDF parser emit an error if the mesh file specified in the submesh element doesn't support submeshes, or doesn't actually contain a submesh of the given name.

patmarion, 2016-03-07:
Ok.  I can propose two options.  There may be there are others ways, let me know what you think.

1) have a preprocessing script to extract submeshes and save them in separate files (using supported file formats) along with the .dae file.  Then drake and the visualizer can look for the submesh files and load them directly.

2) add support for collada (.dae) and obj (.obj) groups to drake and to the visualizer.  One way to do this is to include the assimp third party library as a new dependency:  https://github.com/assimp/assimp

david-german-tri, 2016-03-07:
As a workaround, we started doing (1) last week in TRI projects that use Drake.  I intended this issue to track (2).  I agree replacing Drake's homemade object import code with an external library is a great approach.  I'll work on it as other priorities permit, unless someone else tackles it first.

tri-ltyyu, 2019-03-22:
Use case may be outdated. Please re-open/open new issue if needed

closed 2019-03-22 by tri-ltyyu