RobotLocomotion/drake #1790: How should I tell Drake about Gurobi?

AndreasMegatranus, 2016-03-02:
I've got my academic license for Gurobi and installed it in the Gurobi
default location
(/opt/gurobi650). The license file is in my home directory.

When I run runAtlasBalancing, I still get the "cannot find required
dependency: gurobi" error.
It says something about setting some environment variables. What do I
still need to do to tell
Drake where gurobi is? Do I need to reinstall Drake?

RussTedrake, 2016-03-03:
definitely no need to reinstall drake. 

you can see in https://github.com/RobotLocomotion/drake/blob/master/drake/util/checkDependency.m#L238 that it is just trying to call 'gurobi' from the command line in matlab.  as long as gurobi is in your matlab path, it should find the dependency.  

drake does cache the dependency checks, so if you called checkDependency before gurobi was in your path it will not look on disk for it again.  At worst you can call `clear checkDependency.m` or restart matlab to get past that one.

RussTedrake, 2016-03-05:
please reopen if you are still stuck.

closed 2016-03-05 by RussTedrake