RobotLocomotion/drake #1792: Error in Quadrotor runMixedIntegerForest example

JanaMHartman, 2016-03-02:
In the runMixedIntegerForest example, the script fails with a subscripted assignment dimension mismatch, and also produces a Mosek max time error.  The example runMixedIntegerSimpleForest example runs and produces a trajectory that can be run in the visuliazer, although there is a Mosek stall error in the output.

runMixedIntegerForest Output:
attempting to launch the drake director
Running mixed-integer convex program for 3rd-degree polynomial...

MOSEK Version 7.1.0.45 (Build date: 2016-1-25 06:52:46)
Copyright (c) 1998-2016 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Linux/64-X86

Computer
  Platform               : Linux/64-X86    

Problem
  Name                   :                 
  Objective sense        : min             
  Type                   : CONIC (conic optimization problem)
  Constraints            : 7200            
  Cones                  : 211             
  Scalar variables       : 1424            
  Matrix variables       : 0               
  Integer variables      : 56              

Optimizer started.
MOSEK warning 500: The license expires in 3 days.
MOSEK warning 500: The license expires in 3 days.
Mixed integer optimizer started.
Optimizer  - threads                : 6
BRANCHES RELAXS   ACT_NDS  BEST_INT_OBJ         BEST_RELAX_OBJ       REL_GAP(%)  TIME  
0        1        0        NA                   6.3452378842e-02     NA          0.5   
0        1        1        NA                   6.3452378842e-02     NA          0.6   
0        1        1        NA                   6.3452378842e-02     NA          1.1   
0        1        1        NA                   6.3452378842e-02     NA          1.7   
0        1        1        NA                   6.3452378842e-02     NA          2.3   
0        1        1        NA                   6.3452378842e-02     NA          2.9   
1        2        2        NA                   6.3452378842e-02     NA          3.4   
3        4        4        NA                   6.3452390502e-02     NA          4.1   
7        8        4        NA                   6.3452390844e-02     NA          5.0   
9        10       6        NA                   6.3452390844e-02     NA          5.6   
12       13       7        NA                   6.3452392102e-02     NA          6.4   
16       17       7        NA                   6.3452393473e-02     NA          7.5   
20       21       11       NA                   6.3452393473e-02     NA          8.4   
24       25       13       NA                   6.3452393473e-02     NA          9.4   
33       34       16       NA                   6.3452423195e-02     NA          11.0  
45       46       26       NA                   6.3452428728e-02     NA          13.5  
69       71       40       NA                   6.3452432197e-02     NA          17.7  
105      107      48       NA                   6.3452432197e-02     NA          22.5  
153      155      58       NA                   6.3452432197e-02     NA          25.3  
207      210      62       NA                   6.3452432197e-02     NA          33.3  
267      272      96       NA                   6.3452432197e-02     NA          45.8  
363      374      148      NA                   6.3452449019e-02     NA          61.1  
494      521      173      NA                   6.3452471269e-02     NA          80.2  
646      675      201      NA                   6.3452471269e-02     NA          91.4  
796      836      231      NA                   6.3452483138e-02     NA          102.2 
952      998      233      NA                   6.3452593400e-02     NA          119.6 
1110     1158     269      NA                   6.3452614495e-02     NA          132.8 
1276     1324     303      NA                   6.3452614495e-02     NA          139.2 
1440     1489     331      NA                   6.3452614495e-02     NA          146.3 
1606     1655     339      NA                   6.9689476733e-02     NA          151.1 
1771     1820     350      NA                   8.0828355571e-02     NA          158.6 
1927     1983     350      NA                   8.1084790912e-02     NA          171.7 
2080     2142     359      NA                   8.1085120043e-02     NA          185.6 
2231     2297     382      NA                   8.1119886060e-02     NA          196.3 
2391     2460     396      NA                   1.5120289582e-01     NA          205.1 
2544     2621     415      NA                   3.3154419962e-01     NA          217.1 
2703     2782     448      NA                   3.3163434607e-01     NA          225.6 
2867     2947     454      NA                   3.3216842487e-01     NA          233.4 
3034     3114     451      NA                   3.5860460766e-01     NA          237.6 
3190     3275     421      NA                   5.4687316916e-01     NA          252.1 
3353     3438     412      NA                   5.4687379024e-01     NA          260.8 
[... 136 lines cut ...]
  Optimizer                 -                        time: 0.09    
    Interior-point          - iterations : 6         time: 0.09    
      Basis identification  -                        time: 0.00    
        Primal              - iterations : 0         time: 0.00    
        Dual                - iterations : 0         time: 0.00    
        Clean primal        - iterations : 0         time: 0.00    
        Clean dual          - iterations : 0         time: 0.00    
        Clean primal-dual   - iterations : 0         time: 0.00    
    Simplex                 -                        time: 0.00    
      Primal simplex        - iterations : 0         time: 0.00    
      Dual simplex          - iterations : 0         time: 0.00    
      Primal-dual simplex   - iterations : 0         time: 0.00    
    Mixed integer           - relaxations: 0         time: 0.00    

Elapsed time is 0.344190 seconds.
Subscripted assignment dimension mismatch.

Error in MISOSTrajectoryProblem/solveTrajectory (line 279)
          coeffs(i,k,:) = fliplr(reshape(ct,1,[]));

Error in runMixedIntegerEnvironment (line 100)
ytraj = prob.solveTrajectory(start, goal, safe_regions, safe_region_assignments);

Error in runMixedIntegerForest (line 28)
runMixedIntegerEnvironment(r, start, goal, lb, ub, seeds, degree, n_segments, n_regions);

rdeits, 2016-03-03:
Yeah, it looks like Mosek just ran out of time (which may be a function of your CPU speed or system load) and produced a garbage solution. Can you try increasing the timeout for the Mosek solve, just to make sure that's all that's going on? The line to change is line 254 of drake/examples/Quadrotor/MISOSTrajectoryProblem.m:

```matlab
diagnostics = optimize(constraints, objective, sdpsettings('solver', 'mosek', 'mosek.MSK_DPAR_MIO_TOL_REL_GAP', 1e-2, 'mosek.MSK_DPAR_MIO_MAX_TIME', 600, 'verbose', 3));
```

JanaMHartman, 2016-03-05:
I changed the time from 600 to 1200.  It now tells me the problem is Primal Infeasible.  The objective function value jumps abruptly in the initial optimization.  I've included the beginning and the end when the jump occurs.  The final error, after the 5th degree run, is the same as before since a solution is still not found.

runMixedIntegerForest
Running mixed-integer convex program for 3rd-degree polynomial...

MOSEK Version 7.1.0.45 (Build date: 2016-1-25 06:52:46)
Copyright (c) 1998-2016 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Linux/64-X86

Computer
  Platform               : Linux/64-X86    

Problem
  Name                   :                 
  Objective sense        : min             
  Type                   : CONIC (conic optimization problem)
  Constraints            : 7200            
  Cones                  : 211             
  Scalar variables       : 1424            
  Matrix variables       : 0               
  Integer variables      : 56              

Optimizer started.
MOSEK warning 500: The license expires in 1 days.
MOSEK warning 500: The license expires in 1 days.
Mixed integer optimizer started.
Optimizer  - threads                : 6
BRANCHES RELAXS   ACT_NDS  BEST_INT_OBJ         BEST_RELAX_OBJ       REL_GAP(%)  TIME  
0        1        0        NA                   6.3452378842e-02     NA          0.5   
0        1        1        NA                   6.3452378842e-02     NA          0.6   
0        1        1        NA                   6.3452378842e-02     NA          1.2   
0        1        1        NA                   6.3452378842e-02     NA          1.7   

...

9633     9832     22       NA                   6.6614503408e+01     NA          699.9 
9649     9848     16       NA                   6.6614503408e+01     NA          701.3 
9661     9861     12       NA                   6.6614543163e+01     NA          703.4 
9669     9869     12       NA                   6.6614546845e+01     NA          704.5 
9679     9879     6        NA                   1.0000000000e+300    NA          706.0 
9683     9883     8        NA                   1.0000000000e+300    NA          706.6 
9688     9888     9        NA                   1.0000000000e+300    NA          707.5 
9691     9891     10       NA                   1.0000000000e+300    NA          708.6 
9695     9896     8        NA                   1.0000000000e+300    NA          710.7 
9699     9901     10       NA                   1.0000000000e+300    NA          712.4 
9704     9907     11       NA                   1.0000000000e+300    NA          714.2 
9708     9911     13       NA                   1.0000000000e+300    NA          714.7 
9717     9921     10       NA                   1.0000000000e+300    NA          716.8 
9722     9926     11       NA                   1.0000000000e+300    NA          717.4 
9726     9930     13       NA                   1.0000000000e+300    NA          717.8 
9735     9939     4        NA                   1.0000000000e+300    NA          718.6 
9737     9941     4        NA                   1.0000000000e+300    NA          718.9 
9739     9943     4        NA                   1.0000000000e+300    NA          719.2 
9741     9945     2        NA                   1.0000000000e+300    NA          719.4 
9742     9946     3        NA                   1.0000000000e+300    NA          719.9 
9745     9949     4        NA                   1.0000000000e+300    NA          720.3 
9748     9952     5        NA                   1.0000000000e+300    NA          720.6 
9751     9955     6        NA                   1.0000000000e+300    NA          720.9 
9754     9958     5        NA                   1.0000000000e+300    NA          721.2 
9758     9962     3        NA                   1.0000000000e+300    NA          721.6 
9761     9965     2        NA                   1.0000000000e+300    NA          721.8 
9763     9967     0        NA                   1.0000000000e+300    NA          722.1 

Objective of best integer solution : Not available.
Best objective bound               : 1.000000000000e+300     
Construct solution objective       : Not employed
Construct solution # roundings     : 0
User objective cut value           : 0
Number of cuts generated           : 0
Number of branches                 : 9763
Number of relaxations solved       : 9967
Number of interior point iterations: 179659
Number of simplex iterations       : 0
Time spend presolving the root     : 0.04
Time spend in the heuristic        : 0.00
Time spend in the sub optimizers   : 0.00
  Time spend optimizing the root   : 0.49
Mixed integer optimizer terminated. Time: 722.09

Optimizer terminated. Time: 722.11  

Integer solution solution summary
  Problem status  : PRIMAL_INFEASIBLE
  Solution status : UNKNOWN
  Primal.  obj: 0.0000000000e+00    Viol.  con: 1e+01    var: 0e+00    cones: 0e+00    itg: 0e+00  
Optimizer summary
  Optimizer                 -                        time: 722.11  
    Interior-point          - iterations : 0         time: 0.00    
      Basis identification  -                        time: 0.00    
        Primal              - iterations : 0         time: 0.00    
        Dual                - iterations : 0         time: 0.00    
        Clean primal        - iterations : 0         time: 0.00    
        Clean dual          - iterations : 0         time: 0.00    
        Clean primal-dual   - iterations : 0         time: 0.00    
    Simplex                 -                        time: 0.00    
      Primal simplex        - iterations : 0         time: 0.00    
      Dual simplex          - iterations : 0         time: 0.00    
      Primal-dual simplex   - iterations : 0         time: 0.00    
    Mixed integer           - relaxations: 9967      time: 722.09  

Elapsed time is 724.299104 seconds.
done!

rdeits, 2016-03-07:
Okay, it looks like this test must have broken at some point and was never caught. I'll start bisecting and see if I can figure out what happened. Thanks for reporting!

rdeits, 2016-03-07:
It looks like you uncovered a real bug in https://github.com/rdeits/iris-distro Fix is incoming... Thanks again!

PR #1822 (2016-03-08): update IRIS to fix automatic region seeding
An important option was getting dropped when moving from Matlab to the mex file, which made the automatic regions fail to cover enough of the free space. 

fixes #1792
commit 352a5691 (2016-03-08): update IRIS to fix automatic region seeding

fixes #1792
commit 5344e42b (2016-03-10): update IRIS to fix automatic region seeding

fixes #1792
closed 2016-03-08 by RussTedrake