RobotLocomotion/drake #1803: need a lightweight c++ image loading library

RussTedrake, 2016-03-05:
@crogers3 recently brought in https://github.com/RobotLocomotion/drake/tree/master/drake/thirdParty/cimg , which is almost perfect but is LGPL... so it would infect our license.  

This is needed for e.g. #1616 .  Must work on mac,linux,win.  Any suggestions?

crogers3, 2016-03-05:
Another library I had seen was [lodepng](https://github.com/lvandeve/lodepng), which I had tried out and got to work, but it doesn't seem to have an official licence, just a [small statement in its header file](https://github.com/lvandeve/lodepng/blob/master/lodepng.h#L1-L24). I also am not sure about its cross-platform capability. It worked for me on linux.

I ultimately decided to switch to cimg because it seemed more legitimate in terms of licensing and cross-platform compatibility.

sherm1, 2016-03-05:
Lodepng's [license](https://github.com/lvandeve/lodepng/blob/master/lodepng.h#L1-L24) is non-viral and BSD-like so should be OK. Simbody has shipped with an embedded lodepng for a long time and it works on Linux/Windows/OSX. Is that enough functionality to replace cimg though?

RussTedrake, 2016-03-05:
We aren’t actually using cimg for anything yet. We just brought it in to start
supporting terrain heightmap loading from image. So we can get by with less...

avalenzu, 2016-03-07:
Is reading bitmaps the only required functionality?

avalenzu, 2016-03-07:
Re. LodePNG: More specifically, it uses the [zlib/libpng](https://opensource.org/licenses/Zlib) license, word-for-word. If reading bitmaps is all we need, it seems like a good option - especially given what @sherm1 points out about it having been tested cross-platform through Simbody.

avalenzu, 2016-03-08:
@RussTedrake, any more thoughts on requirements?

RussTedrake, 2016-03-09:
I think it’s totally reasonable to say that any image loaded as a heightmap into
drake must be converted to a png before load. Seems like loadpng should do the
trick.

closed 2016-05-08 by RussTedrake