RobotLocomotion/drake #1804: bot2-lcmgl not found

zeber19, 2016-03-06:
Hi I am running into the following error when attempting to make from source on linux (ubuntu 14.04)

`[ 87%] Performing configure step for 'drake'
-- CMAKE_INSTALL_PREFIX=/home/masher/git/drake-distro/build
CMake Warning at cmake/mex.cmake:241 (message):
  Your MEX compiler flags contained '-ansi', but we've removed that flag for
  compatibility with C++11
Call Stack (most recent call first):
  CMakeLists.txt:120 (mex_setup)

-- Found matlab
CMake Error at CMakeLists.txt:178 (message):
  

  

  ----- ERROR: bot2-lcmgl not found

-- Configuring incomplete, errors occurred!
`

I have been tracking down missing dependencies but am seemingly out of ideas at this point. How should I proceed?

Thanks

RussTedrake, 2016-03-06:
My guess is it's this #1774

zeber19, 2016-03-06:
Yes, that seemed to be the problem. Thanks much

closed 2016-03-06 by zeber19