RobotLocomotion/drake #1805: Need a standardized copyright message in every source file

sherm1, 2016-03-06:
Each file in an open source project should start with a standardized copyright/license block. Users can typically do anything with the code *except* remove the block.

We need to decide:
- what license to use (I favor Apache 2.0)
- who should hold the copyright (that determines who must agree on a change to the license)
- what other information (if any) should be included (e.g. credit to funding agencies & contributors)
- how the header should be formatted

and then figure out how to update all the existing files.

Here are a couple of examples to get started: 
- [Google tensorflow](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/core/graph/equal_graph_def.h#L1-L14) (minimalist: no credit or copyright holding for anyone but Google); OSRF uses this format also
- [Simbody](https://github.com/simbody/simbody/blob/master/SimTKcommon/Mechanics/include/SimTKcommon/internal/Rotation.h#L4-L25) (flashy: credits funding agency, provides authors copyright control, credits contributors)

RussTedrake, 2016-03-06:
thanks!  i would err on the side of the minimal.

liangfok, 2016-03-06:
We can use a bash script to update all existing source files in a directory with the copyright text: 

http://unix.stackexchange.com/questions/125421/how-to-prepend-a-particular-text-file-contents-to-every-text-file-in-a-directory

Ideally, the git commit should be done by a robot to make it clear the change was automated.

jamiesnape, 2016-03-07:
Resolving this would keep cpplint happy too.

jamiesnape, 2016-03-15:
I would like to fix the hundreds of lint errors due to this, so I propose the following.

C++ / Java:
```
// Copyright 2016 Robot Locomotion Group @ CSAIL. All rights reserved.
//
// All components of Drake are licensed under the BSD 3-Clause License.
// See LICENSE.TXT or http://drake.mit.edu/ for details.
```

CMake / Python / Shell:
```
# Copyright 2016 Robot Locomotion Group @ CSAIL. All rights reserved.
#
# All components of Drake are licensed under the BSD 3-Clause License.
# See LICENSE.TXT or http://drake.mit.edu/ for details.
```

MATLAB:
```
% Copyright 2016 Robot Locomotion Group @ CSAIL. All rights reserved.
%
% All components of Drake are licensed under the BSD 3-Clause License.
% See LICENSE.TXT or http://drake.mit.edu/ for details.
```
The year is the year the file was created. The template that the linter follows is `Copyright YEAR ORGANIZATION`.

RussTedrake, 2016-03-15:
@jamiesnape -- we've decided to step back and resolve the copyright correctly before deciding.  And I believe you can disable the errors with a one line argument to the lint exe.  @david-german-tri can you remind me what that was?

jamiesnape, 2016-03-15:
Sure, I can can disable it, if you prefer. I know the command.

sherm1, 2016-03-15:
> we've decided to step back and resolve the copyright correctly before deciding

Just so it won't get forgotten -- both MIT and TRI are likely to be the copyright holders.

david-german-tri, 2016-03-15:
For anyone else following along, the flag Russ mentioned is

`cpplint --filter="-legal/copyright"`

RussTedrake, 2016-05-25:
offline discussion is in progress.

jwnimmer-tri, 2018-08-14:
No, we still need to add some TBD boilerplate atop every file.  It's still on my TODO list.  At this point, it is more difficult because will probably need to `git blame` our way through the list of copyright holders for each file.

jwnimmer-tri, 2019-11-06:
https://softwarefreedom.org/resources/2012/ManagingCopyrightInformation.html has some good advice.

jamiesnape, 2019-11-07:
I think I like their suggestions on centralized licenses and their suggested boilerplate more or less (though I would not be bothered if we skipped the boilerplate). I think individual attribution in files is probably barely viable (or worth the effort) now, so if we need (corporate/institutional) names for copyright purposes beyond source control, an overall `AUTHORS` file is going to be easiest (see Bazel for one example, of course).

I would also like to see the `LICENSE` file be a verbatim dump of the BSD license. We are mixing legal text with advisory, non-legal text at the moment. If we need notes on exceptions, they should be in `third_party`, `tools/workspace`, in line in the relevant files, or in a `NOTICES file`, I think.

jamiesnape, 2020-08-07:
Since I have been running automated license checking tools through our code and our dependencies code the last few days, I would just say that having something, anything, is important otherwise it needs manual inspection, which is not practical in a project the size of Drake, especially as we have third-party code in tree.

At a minimum you want on one line the word `Copyright`, the year of creation, and some entity even if it is "The Drake Development Team". Assuming we stick with BSD 3-clause, on a different line, you would want the words "BSD 3-clause license", or "BSD-3-Clause", or "BSD-3-clause". These should be the first lines in the file, modulo shebangs or codelines Whatever is chosen, running the Debain license checker through the tree and checking the license and copyright fields are identified correctly is a good idea.

jwnimmer-tri, 2026-04-15:
I still plan to do it.