RobotLocomotion/drake #1808: Consider adding a CONTRIBUTING.md file

sherm1, 2016-03-06:
GitHub convention is for a repo to have a top-level file called `CONTRIBUTING.md` that provides guidance to people who want to contribute to the software. GitHub understands that convention and points potential contributors there automatically.

Legally, the most important part of this is the "Contributor License Agreement" (CLA) that stops people from doing evil/ignorant things like slip in GPL code or patented material. This has to be balanced with a friendly tone that encourages contribution. Here are a few examples:
- [Google Tensorflow](https://github.com/tensorflow/tensorflow/blob/master/CONTRIBUTING.md) (just the CLA)
- [Simbody](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md) (attempts to be friendly and includes coding conventions and contributor list)
- [NumPy](https://github.com/numpy/numpy/blob/master/CONTRIBUTING.md) (short, with links to coding conventions and procedures)

RussTedrake, 2016-03-06:
yikes.  love the contributing.md.  not a fan at all of the CLA.  I think we can rely on the code review to make sure people don't sneak in GPL'd code... no?  Putting a message there saying that they take responsibility is one thing, but forcing people to sign is too much...

sherm1, 2016-03-06:
I agree that making people sign a CLA is way overkill -- Google does it but most other projects don't. There is a huge amount of internet chatter about this which we waded through for Simbody & OpenSim. We concluded that you have to address the issue (PR review is not enough because people need to actively grant rights), but that a "shrink wrap" approach like the one in the Simbody file is good enough.

sherm1, 2016-03-06:
This is the [Simbody CLA](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md#contributor-license-agreement). No one has to sign it -- we just say at the start of the file that they agree to it by contributing:

> Important note: Simbody is an open source project licensed under extremely flexible terms intended to encourage use by anyone, for any purpose. When you make a contribution to the Simbody project, you are agreeing to do so under those same terms. The details are below; if you aren't comfortable with those terms, we're still friends but you shouldn't contribute.

Most of these licensing agreements have never been tested in court, but I think it is critical to be explicit about what we expect of contributors -- most people want to do the right thing but many (including potential PR reviewers) might not be aware that a single line of GPL could be death for a real open source project!

RussTedrake, 2016-03-06:
That text seems very reasonable to me.

PR #1809 (2016-03-06): added very simple initial CONTRIBUTING.md.  
resolves #1808

@sherm1 -- this is super minimal.  but perhaps good enough to start?
commit 5d1431c4 (2016-03-06): added very simple initial CONTRIBUTING.md.  resolves #1808
closed 2016-03-06 by RussTedrake