RobotLocomotion/drake #1811: PRs should build documentation, too

RussTedrake, 2016-03-06:
The build servers are not actually building doxygen or sphinx documentation.  Bad syntax in a commit could bring down our hosted documentation (which is now our front page).

jamiesnape, 2016-03-07:
You should also set the documentation to update during a successful nightly build. This also minimizes your risks.

sherm1, 2016-03-15:
I would like to be able to test the doxygen I write along with code changes in my local build before I commit them. It would be nice to have a build target to allow `make doxygen` or building docs from within the IDE (is this worthy of a separate issue?). Meanwhile, is there an easy way to build doxygen locally (and apologies if I missed obvious instructions somewhere)?

RussTedrake, 2016-03-16:
`make doc` in drake-distro/drake 
it was the logical place for it when drake was a module.  now feels illogical in the superbuild format.  and would be great to push it into cmake.

sherm1, 2016-03-16:
Opened issue #1870 requesting a doxygen target.

jamiesnape, 2016-04-21:
Depends on #2143.

PR #2143 (2016-05-05): Add CMake target for build documentation (again)
Originally reviewed in #2128. Reverted in #2140. Closes #1811. Closes #1870.

Includes nasty hack from http://stackoverflow.com/questions/27979803/external-relative-link-in-sphinx-toctree-directive, unfortunately.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2143)
closed 2016-05-05 by jamiesnape