RobotLocomotion/drake #1817: bot2-lcmgl not found

RoboEngineer, 2016-03-07:
Hi all,

I installed Java but still got an error on "bot2-lcmgl not found" with the following errors.
"-- Configuring incomplete, errors occurred!
See also "/Users/Y/drake-distro/drake/pod-build/CMakeFiles/CMakeOutput.log".
make[3]: *** [drake-prefix/src/drake-stamp/drake-configure] Error 1
make[2]: *** [CMakeFiles/drake.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2"

Could you please give a little bit more detail instruction on how to solve it? Thank you very much!

RussTedrake, 2016-03-07:
see #1774

RoboEngineer, 2016-03-07:
Thanks Russ! But I am not sure how to check for java at the drake-distro level and error out immediately (before they get to compiling libbot)? Do you think you could provide a more detail instruction? Really appreciate!

jonathan-owens, 2016-03-07:
@RoboEngineer Hey Robo -- Try running ``make clean`` in your super-build directory, then clean your CMake cache and reconfigure and regenerate and then rebuild drake. 

I am working on a fix right now so this doesn't happen while still allowing one to specify which Java in the super-build and then have that propagate to the external projects.

RoboEngineer, 2016-03-07:
@jonathan-owens Thank you very much Jonathan. I got some error when I just type in `make clean`. But I can clean it with 'rm -rf build'. 
And I redid the steps starting from "Mandatory platform specific instructions" then it works!

Thank you very much!

jonathan-owens, 2016-03-07:
Yes, the ``make clean`` has some errors, that is expected. That's being addressed. See, e.g., issue #1338. Glad you're up and working!

@RussTedrake should we close this?

closed 2016-03-07 by RoboEngineer