RobotLocomotion/drake #1835: pathlib files snuck into repo

RussTedrake, 2016-03-09:
the win64, linux64, and mac64 directories in drake-distro root should not be there.   

must have come in through merging from @psiorx .  I had pushed them in by accident, then force pushed a revert to get them out.  @psiorx must have merged inbetween those changes.  my bad.  it happened here:
https://github.com/RobotLocomotion/drake/commit/c809ce219fc063c4d8a5ce06c8aa5f2ae24e476f

They include binaries, and definitely should not be here, not even in the history.  @rdeits -- would you be willing to help me very carefully excise them from the git history?

rdeits, 2016-03-09:
Sure, I'm looking into it right now.

rdeits, 2016-03-09:
Okay, I just pushed rdeits/rd-purge-path, which you can use to replace RobotLocomotion/master. It has all the current commits, but with those three folders purged. 

Any branches made before that commit came in will be unaffected, so most of our branches should be fine. Any branches that do include the bad commit will have to be rebased, but it's a one-liner. If any of the pull requests try to bring the bad commit back in, I can help them rebase it out.

rdeits, 2016-03-10:
Okay, this was fixed by a force-push, which meant rewriting the history since Sunday. That will confuse git, but it's pretty easy to fix. 

If you have done a `git pull` since Sunday, then you should do the following to fix your master branch:

```
git checkout -b old-master-backup # make a backup branch in case something goes wrong
git checkout master
git fetch origin
git reset --hard origin/master
```

If everything looks good and you're happy, you can delete the backup branch with:

```
git branch -D old-master-backup
```

If you have your own branch, it's *probably* fine, as long as you made the branch before Sunday and haven't done a `git merge master` since then. But if your branch has the `win64`, `linux64`, and `mac64` directories in it, then you'll need to clean it up. The fix involves using `git rebase` with the `--onto` flag, but the exact step will depend on your situation. We can help with specific situations.

closed 2016-03-10 by rdeits

david-german-tri, 2016-03-10:
"Any branches that do include the bad commit will have to be rebased, but it's a one-liner."

Could you please provide the one-liner?

patmarion, 2016-03-10:
git checkout <the branch>

    # run git log to find the commit *before* your first commit on the branch

    git rebase --onto master <sha1 of commit before your first commit>

So for your PR, it looks like the base commit is 4cf81e7, if that is correct then just insert that into the rebase command.