RobotLocomotion/drake #1837: Writing link information to a file

amcastro-tri, 2016-03-09:
I personally would find very useful to have a file per link where positions, velocities and accelerations are written. Then for quantitative comparisons I can make plots in either Matlab, octave or whatever tool of my choice.

Something like having a method RigidBodyTree::writeLinksState(Scalar time). This would write files: link1->name.dat, link2->name.dat, ... etc.

This file would have columns with the information:
time x y z roll pitch yaw vx vy vz wx wy wz vdotx vdoty vdotz wdotx wdoty wdotz

Maybe even include the quaternion.

Is there such a thing implemented in Drake already? if not, what would the best place where to start to get this information?

RussTedrake, 2016-03-09:
As we discussed in person, in matlab a call to simulate will return a trajectory object which has the state record but also the information necessary to correctly interpolate between timesteps.  (It also has many methods, and can be used as a system in and of itself).

The goal is to implement this in c++.  Has not happened yet.

amcastro-tri, 2016-03-10:
I see, thank you. I was just trying to learn what methods were already available to quantitatively analyze the solution. I am still learning where all variables are and with that I'll try to output them into something I can open with either octave or even ParaView.

liangfok, 2016-03-10:
Once [ROS](http://www.ros.org/) is integrated with Drake, we can start publishing [sensor_msgs/JointState](http://docs.ros.org/jade/api/sensor_msgs/html/msg/JointState.html) and [tf](http://wiki.ros.org/tf) messages onto ROS topics that we can then save and replay using [rosbag](http://wiki.ros.org/rosbag). 

rosbag is just a data capture / storage tool. To get the data into octave or Matlab, we can write a simple python script that subscribes to the topics of interest and saves it to a plain text comma separated file.

The benefit of using rosbag is it can easily replay the data on the ROS topics as if the robot(s) were running for real.

RussTedrake, 2016-03-10:
fwiw - we already have that capability in drake-distro/build/bin/lcm-logger and drake-distro/build/bin/lcm-logplayer-gui (or lcm-logplayer)

iamwolf, 2016-03-10:
@liangfok @amcastro-tri Once you have a lcmlog you can use [bot-log2mat](https://github.com/RobotLocomotion/libbot/blob/94fe5290329646959e1c27896d529abba81e6d93/bot2-lcm-utils/python/src/bot_log2mat/log_to_mat.py) to create Matlab files for you to investigate. It's most likely already on your systems - source the drake build folder and then type ``bot-log2mat``

amcastro-tri, 2016-03-10:
Thank you @RussTedrake. I'm sorry I didn't pass this information along when you told me about it yesterday. I tried it out and works. 
@iamwolf, bot-log2mat is a Python script that requires the package python-scipy to be installed. 

After running bot-log2mat and loading the output into [Octave](https://www.gnu.org/software/octave/) I get a data structure with the arrays:

- DRAKE_VIEWER_DRAW
- DRAKE_VIEWER_LOAD_ROBOT
- DRAKE_VIEWER_STATUS

What variables are stored in these arrays?

liangfok, 2016-03-10:
DRAKE_VIEWER_DRAW contains position and orientation of every link in the robot. When running the car demo, it's transmitted at anywhere from 50-100Hz on my computer. I'm still trying to figure out how to easily view the current angle of every joint in the robot.

RussTedrake, 2016-03-11:
seems like this has been adequately addressed.

closed 2016-03-11 by RussTedrake