RobotLocomotion/drake #1847: How can I visualize control signals for biped robot

alexbuyval, 2016-03-11:
Hi

I am trying to use the Drake software for ['Darwin' robot](http://www.trossenrobotics.com/p/darwin-OP-Deluxe-humanoid-robot.aspx). I used 'examples/Atlas' folder as a start point, then I have created a suitable urdf model of Darwin and set his parameters. 
However, I faced out that when Darwin is trying to do the first step, it seems that 'r_leg_kny' joint rotate another direction.
[image]
I believe that surface of the foot should be parallel floor.
I would like to debug control joint values and compare them with Atlas control values. But i don't how can I do it.
I found in code a mention about lcmLogger. I have added '.*' in capture_lcm_channels, but Matlab crash after this each time. 
Is the lcmLogger what I need? If so, how can I use it correctly? 
Maybe there is another way to debug/visualize control joint values?

Best Regards,
Alex

RussTedrake, 2016-03-11:
There are many ways.  The output of simulate in matlab gives a signal trajectory back as (optional) outputs.

We also use the signal_scope tool for visualizing and debugging signals sent over lcm.   @amcastro -- that might be useful to you as well.

If you tell us more about how you are running your sim we might be able to help more.

alexbuyval, 2016-03-11:
Thank you @RussTedrake for your quick answer.

I am running the simulation the same way as in runAtlasWalking function. 
I found the line in simulateWalking.m file:
`[ytraj] = simulate(sys, [0, T], x0, options);`
and have added 'lcmlog' there:
`[ytraj,xtraj,lcmlog] = simulate(sys, [0, T], x0, options);`

Also I have added the following line:
`options = applyDefaults(options, struct('capture_lcm_channels', '.*'));`
to capture all LCM messages. However, Matlab is crashed after this.  I guess that I should add a certain LCM channel to capture control signals, but I don't know which.
Also I don't know how I can analyze lcmlog after the simulation was finished.

Best Regards,
Alex

RussTedrake, 2016-03-11:
it should not crash. please send that error if you can capture it.
if you are ok plotting in matlab, then simply example ytraj and xtraj. you 
can
call fnplt on it. and plot specific elements of the vector.
if you want to use the signal scope, then you can simply run it during the
initial simulation. if you want to publish the lcm traffic again during
playback, then capture lcmlog and pass it to the playback methods of 
visualizer.

alexbuyval, 2016-03-11:
There are no any specific information about what is a cause of crash.
[image]
It happens also if I add the same 'capture_lcm_channels' option in Atlas example.

alexbuyval, 2016-03-11:
I have just found the output about crash in terminal:

```
alex@alex-ubuntu:~$ matlab 

(process:1632): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set
LCM self test failed!!
Check your routing tables and firewall settings

(process:1632): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set
LCM self test failed!!
Check your routing tables and firewall settings

(process:1632): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set
LCM self test failed!!
Check your routing tables and firewall settings

(process:1632): GLib-WARNING **: unknown option bit(s) set
lcm_subscribe: Error while compiling regular expression ^LCM_SELF_TEST$ at char 0: unknown option bit(s) set
LCM self test failed!!
Check your routing tables and firewall settings
*** buffer overflow detected ***: /home/alex/MATLAB/R2015b/bin/glnxa64/MATLAB terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x78c4e)[0x7efd4ad28c4e]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7efd4adc8e8c]
/lib/x86_64-linux-gnu/libc.so.6(+0x116e80)[0x7efd4adc6e80]
/lib/x86_64-linux-gnu/libc.so.6(+0x118dd7)[0x7efd4adc8dd7]
/home/alex/drake-distro/drake/util/lcmLogger.mexa64(+0x2b8d)[0x7efc12f62b8d]
/home/alex/drake-distro/drake/util/lcmLogger.mexa64(mexFunction+0x632)[0x7efc12f66552]

```

RussTedrake, 2016-03-11:
I believe that’s the old symlink issue: https://github.com/RobotLocomotion/drake/issues/960
- Russ

alexbuyval, 2016-03-11:
> I believe that’s the old symlink issue: #960

I am not sure. Because I have updated symlinks already according to [this instruction.](http://drake002.csail.mit.edu/drake/sphinx/ubuntu.html)

- Alex

patmarion, 2016-03-13:
That looks like it could be the lcm pcre issue-

https://github.com/openhumanoids/oh-distro-private/issues/644#issuecomment-97905467

I think it was resolved by using Matlab version 2014a.  2014b and newer may not be compatible due to the lcm pcre issue.

To answer the original question- if you are publishing signals with LCM, then you can use the bot-spy tool to view the signals.  If you have already used bot-spy and are looking for some extra features for plotting LCM signals, see signal-scope: https://github.com/openhumanoids/signal-scope

Signal scope is an optional extra provided by the drake build.  Run "make options" and enable WITH_SIGNALSCOPE.

alexbuyval, 2016-03-15:
@patmarion 
Thank you! I have just installed the MATLAB 2014a version and it works now.

I tried to use the bot-spy. It works, however I see only four channels there, when Atlas's walking simulation is running. Is it okay?
[image]

I have found link positions in DRAKE_VIEW_DRAW channel, but I need control signals too.

alexbuyval, 2016-03-18:
Hi

I have found that many examples contain calls like this 
`obj = setInputFrame(obj,PendulumInput);`
As I understood the PendulumInput class is inherited from LCMCoordinateFrame class which has 'publish' method. I have set a breakpoint in this method and then to run 'runLQR' in examples/Pendulum folder. However I don't see any lcm messages in bot-spy and the Matlab was not stopped on my breakpoint in publish method.

Please give me advise, what is right way to use LCMCoordinateFrame class to debug?

Best Regards,
Alex

RussTedrake, 2016-03-18:
The LCM coordinate frames will only publish/recieve if you are using “runLCM” instead of “simulate”. By design, they act like a normal block if called using simulate.
My guess (without looking), is that you could modify the line in runLQR to call runLCM instead of simulate (they take the same arguments) and see the behavior you expect

alexbuyval, 2016-03-18:
@RussTedrake 
Thank you! It works now and I have got messages in bot-spy.
However, if I use runLCM, I cannot get output values like this:
`[ytraj,xtraj,lcmlog] = simulate(sys, [0, T], x0, options);`
Am I right?

RussTedrake, 2016-03-18:
Not with the standard interface. (it would be easy to make a local 
modification to runLCM to return those values).

alexbuyval, 2016-03-18:
Thank you!

I have added additional code to get results into runLCM, however I faced out that I cannot get the 'yout' from 'simout', despite of that I have added the following lines into runLCM:

```
    pstruct.SaveOutput = 'on';
    pstruct.OutputSaveName = 'yout';
```

Finally, I have added in the 'output' method of BipedPlanEvalAndControlSystem class the following code to solve my issue:

```
        msg = drake.lcmt_drake_signal();
        temp = obj.control.robot.getJointNames();      
        msg.timestamp = t;
        msg.val = y;
        msg.coord = temp(3:26);
        msg.dim = length(msg.val);
        obj.lc.publish('CONTROL_OUTPUT', msg);
```

Best Regards,
Alex

closed 2016-03-18 by alexbuyval

VizXu, 2016-03-18:
thank you for your reminder

RussTedrake, 2016-03-18:
runLCM calls simulate. Just take the return values of simulate and pass them back out of runLCM… no?