RobotLocomotion/drake #1848: drake005 fails on a PR2 example with "Not a binary MAT-file"

RussTedrake, 2016-03-11:
examples/PR2/runSaggitalPassive (unknown issue)
```
error_id = MATLAB:load:notBinaryFile
error_message = Unable to read MAT-file /Users/russt/.matlab/R2015a/matlabprefs.mat. Not a binary MAT-file. Try LOAD -ASCII to read as text.
is_segfault = 0
```
```Matlab

                            < M A T L A B (R) >
                  Copyright 1984-2015 The MathWorks, Inc.
                   R2015a (8.5.0.197613) 64-bit (maci64)
                             February 12, 2015

 
To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
 

	Academic License

 Calling addpath_gurobi
[[NON-XML-CHAR-0x8]Warning: Unable to read MAT-file /Users/russt/.matlab/R2015a/matlabprefs.mat.
Not a binary MAT-file. Try LOAD -ASCII to read as text.][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]> In slCustomizer/callAllMethods (line 44)
  In slCustomizer/refresh (line 73)
  In slCustomizer.staticRefresh (line 23)
  In sl_refresh_customizations (line 18)
  In slcustomize (line 4)
  In simulinkrc (line 48)
  In DrakeSystem/getModel (line 225)
  In DynamicalSystem/setSimulinkParam (line 497)
  In runSaggitalPassive (line 15)
  In fevalPackageSafe (line 21)][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]Warning: An error occurred trying to load the Simulink HMI engine.  Possible
reasons include pressing Ctrl+C or running MATLAB with the -nojvm option. Close
and re-open any open any Simulink models to assure proper functioning.  For
proper functioning of Simulink HMI and the Simulation Data Inspector, the JVM
is required.][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]> In slxPackager/registerHandler (line 30)
  In slxPackager/executeLoadCallbacks (line 50)
  In slprivate (line 11)
  In load_system (line 21)
  In DrakeSystem/getModel (line 229)
  In DynamicalSystem/setSimulinkParam (line 497)
  In runSaggitalPassive (line 15)
  In fevalPackageSafe (line 21)][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]Warning: An error occurred trying to load the Simulink HMI engine.  Possible
reasons include pressing Ctrl+C or running MATLAB with the -nojvm option. Close
and re-open any open any Simulink models to assure proper functioning.  For
proper functioning of Simulink HMI and the Simulation Data Inspector, the JVM
is required.][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]> In slxPackager/partDefHmi (line 17)
  In slxPackager/registerHandler (line 30)
  In slxPackager/executeLoadCallbacks (line 50)
  In slprivate (line 11)
  In load_system (line 21)
  In DrakeSystem/getModel (line 229)
  In DynamicalSystem/setSimulinkParam (line 497)
  In runSaggitalPassive (line 15)
  In fevalPackageSafe (line 21)][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]Warning: Error using slxPackager/partDefHmi (line 18)
Java exception occurred:
org.apache.commons.lang.NotImplementedException: Code is not implemented
	at
        com.mathworks.messageservice.MessageServiceFactory.getBuilder(MessageServiceFactory.java:42)
        	at
        com.mathworks.messageservice.MessageServiceFactory.getMessageServiceOpaque(MessageServiceFactory.java:23)
        	at
        com.mathworks.toolbox.slobserver.bridge.ServerPubSub.initialize(ServerPubSub.java:30)
        	at
        com.mathworks.toolbox.slobserver.bridge.ServerPubSub.initializeSubscribers(ServerPubSub.java:41)
        
Error in slxPackager/registerHandler (line 30)

Error in slxPackager/executeLoadCallbacks (line 50)

Error in slprivate (line 11)
   [varargout{1:nargout}] = feval(function_name, varargin{1:end});

Error in load_system (line 21)
open_system(sys, 'loadonly');

Error in DrakeSystem/getModel (line 229)
      load_system('simulink');

Error in DynamicalSystem/setSimulinkParam (line 497)
          set_param(obj.getModel(),varargin{i},varargin{i+1});

Error in runSaggitalPassive (line 15)
r = setSimulinkParam(r,'MinStep','0.001');

Error in fevalPackageSafe (line 21)
  feval(f);
][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]> In slxPackager/registerHandler (line 37)
  In slxPackager/executeLoadCallbacks (line 50)
  In slprivate (line 11)
  In load_system (line 21)
  In DrakeSystem/getModel (line 229)
  In DynamicalSystem/setSimulinkParam (line 497)
  In runSaggitalPassive (line 15)
  In fevalPackageSafe (line 21)][NON-XML-CHAR-0x8] 
Error using load
Unable to read MAT-file /Users/russt/.matlab/R2015a/matlabprefs.mat. Not a binary MAT-file. Try LOAD -ASCII to read as text.

Error in prefutils>loadPrefs (line 42)
  fileContents = load(prefFile);

Error in prefutils (line 10)
[varargout{1:nargout}] = feval(varargin{:});

Error in getpref (line 50)
Preferences = prefutils('loadPrefs');

Error in vrgetpref (line 323)
edpref = getpref('VirtualReality', 'Editor', defaults.Editor);

Error in vrclimex_init>phase1 (line 54)
env.preferences = vrgetpref;

Error in vrclimex_init (line 12)
env = feval(sprintf('phase%d', phase));

Error in vrworld (line 88)
  [previd, editFlag] = vrsfunc('VRT3SceneByFilename', foundname);

Error in stl2vrml (line 75)
  w = vrworld('');

Error in RigidBodyMesh/convertToWRL (line 74)
          stl2vrml(fullfile(path,[name,'.stl']),path);

Error in RigidBodyMesh/loadFile (line 30)
      obj = convertToWRL(obj);

Error in RigidBodyMesh/getPoints (line 67)
      pts = loadFile(obj);

Error in RigidBodyGeometry/getPatchData (line 56)
      pts = getPoints(obj);

Error in PlanarRigidBodyVisualizer/updateManipulator (line 41)
          [obj.body(i).x{j},obj.body(i).y{j},obj.body(i).z{j},obj.body(i).c{j}] = getPatchData(b.visual_geometry{j},obj.Tview(:,1),obj.Tview(:,2), obj.Tview(:,3));

Error in PlanarRigidBodyVisualizer (line 24)
      obj = updateManipulator(obj,manip);

Error in PlanarRigidBodyManipulator/constructVisualizer (line 134)
        v = PlanarRigidBodyVisualizer(obj);

Error in TimeSteppingRigidBodyManipulator/constructVisualizer (line 1097)
      v = constructVisualizer(obj.manip,varargin{:});

Error in runSaggitalPassive (line 16)
v = r.constructVisualizer;

Error in fevalPackageSafe (line 21)
  feval(f);

 
<test_name>examples/PR2/runSaggitalPassive</test_name> <error_id>MATLAB:load:notBinaryFile</error_id> <error_message>Unable to read MAT-file /Users/russt/.matlab/R2015a/matlabprefs.mat. Not a binary MAT-file. Try LOAD -ASCII to read as text.</error_message> 
matlab exit code: 1

```

<matlab_test>strcmp(test_name,'examples/PR2/runSaggitalPassive') && strcmp(error_id,'MATLAB:load:notBinaryFile')</matlab_test>

RussTedrake, 2016-04-03:
just tried deleting the matlabprefs.mat file (which was empty) on drake005, per the suggestion here:
http://stackoverflow.com/questions/31926760/how-to-solve-this-error-with-matlabprefs-mat

avalenzu, 2016-04-03:
Did this resolve the issue?

RussTedrake, 2016-04-03:
yes.  it seems to have resolved it.  definitely a good one for TMW.  :)

closed 2016-04-03 by RussTedrake