RobotLocomotion/drake #1853: PlanarRigidBodyVisualizer does not support RigidBodyHeightMapTerrain

zamzami, 2016-03-12:
I would like to change the flat terrain geometry to inclined, step or any other form with obstacles. To my knowledge, there are currently two methods to achieve this in Drake. That is either by defining a new terrain object in the form of Height Map `RigidBodyHeightMapTerrain` or to add the terrain as a separate robot using `addRobotFromURDF`. Since I don't use Bullet anyway I think defining a terrain geometry directly in drake is much faster and more accurate than checking contact points against URDF. (Please correct me if they are exactly equivalent)

Unfortunately,  `PlanarRigidBodyVisualizer` isn't yet ready to host `RigidBodyHeightMapTerrain`. Here is a terrain with a step generated by
 `RigidBodyHeightMapTerrain([-5,1,1,2,2,5],[-1,1],0.5*[0 0 1 1 0 0;0 0 1 1 0 0]);`

Viewed in RigidBodyVisualizer
[image]

and the distorted version viewed in PlanarRigidBodyVisualizer
[image]

I think the source of distortion is due to the way `getPatchData` describes the points of the projected geometry. There is probably a quick fix , but i couldn't investigate further. I also had to transpose the colour array to  `obj.c = hex2dec({'ee','cb','ad'})'/256;`

RussTedrake, 2016-03-12:
The heightmap terrain uses barycentric interpolation.  To avoid the ramp, you should just add two mesh points (in x?) very close to each other - one at the bottom of the step and one at the top of the step (displaced by say 1mm).

But perhaps even easier would be to just make a set of boxes which you add to the bullet world (eg via addCollisionGeometry), like in the prius ramp example.

zamzami, 2016-03-13:
I think it is a problem of visualization (`PlanarRigidBodyVisualizer`) and not in the definition of the height map terrain itself `RigidBodyHeightMapTerrain` For two reasons;
 First, I did exactly as you said (without the little slope though) by defining two heights (0 & 1) at a repeated x value and it is displayed correctly in the 3D `RigidBodyVisualizer`
`RigidBodyHeightMapTerrain([-5,1,1,2,2,5],[-1,1],0.5*[0 0 1 1 0 0;0 0 1 1 0 0]);`

Second, by using the `inspector` tool the robot collides only with the original step and doesn't see the ramp at all. 

Unlike the pirus example, I don't use Bullet nor time stepping. I rely mainly on `ContactImplicitTrajectoryOptimization`

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri