RobotLocomotion/drake #1856: Model Correctness Depends on Order of Elements in SDF

liangfok, 2016-03-13:
The correctness of the model used by Drake depends on the order of the elements in the SDF file. This should not be the case. The existence proof of this issue is given by [prius_v1.sdf](https://github.com/liangfok/drake/blob/087232127f4efc6affa5559d8b8be83afaf4eb93/drake/examples/Cars/prius/prius_v1.sdf) and [prius_v1_reordered.sdf](https://github.com/liangfok/drake/blob/087232127f4efc6affa5559d8b8be83afaf4eb93/drake/examples/Cars/prius/prius_v1_reordered.sdf).

prius_v1.sdf loads fine as shown below (the body and chassis_floor are hidden to not obstruct the front steering linkages):

[image]

prius_v1_reordered.sdf does not load correctly:

[image]

I suspect the problem may be due to child links (`left_hub` and `right_hub`) being ordered before their parent links (`left_tie_rod_arm` and `right_tie_rod_arm`) in the SDF file.

RussTedrake, 2016-03-13:
Off the top of my head, I can think of a few things in the parser that might cause this.  Question: is it legal for a joint tag to come before the link tag that it references?

liangfok, 2016-03-13:
Looking at SDF's [model.xsd](http://sdformat.org/schemas/model.xsd) (which I'm not sure is official), links and joints can appear in any order within the SDF file. However, a cursory look through [Gazebo's models](https://bitbucket.org/osrf/gazebo_models) indicates that it is customary to first define the links followed by the joints.

RussTedrake, 2016-03-13:
what about in your failing test case?

liangfok, 2016-03-13:
In my failing test case, all links are defined prior to joints, so I believe it is a valid SDF file. I still need to investigate this further.

david-german-tri, 2016-05-24:
@liangfok is this still open?  Any thoughts on how to prioritize it?

liangfok, 2016-05-25:
I believe it is still open. I would rank it as low priority since it is not blocking anything.

liangfok, 2017-04-12:
I'm going to close this issue for now but tag #4337 as something to unit test after the SDF parser is swapped.

closed 2017-04-12 by liangfok