RobotLocomotion/drake #1864: RigidBody::getJoint() Fails To Throw Exception If Joint Not Set

liangfok, 2016-03-15:
This problem can be reproduced by trying to load a URDF model containing a transmission for a joint that does not exist. Instead of throwing an exception, it segfaults while trying to access member variable `joint`.

liangfok, 2016-03-15:
Actually, according to the [Google C++ Style Guide](https://google.github.io/styleguide/cppguide.html#Exceptions), we should not be using exceptions. Perhaps `RigidBody::getJoint()` should be refactored to return an error code or simply `false` if the joint does not exist.

sherm1, 2016-03-15:
> Actually, according to the Google C++ Style Guide, we should not be using exceptions. 

Google did that because exceptions didn't work properly in early C++ compilers. That has been fixed but Google is stuck with that rule now. I don't think anyone but Google takes that seriously. I believe the Google style guide already forbids many practices that are standard in Drake already. See [this rant](https://www.linkedin.com/pulse/20140503193653-3046051-why-google-style-guide-for-c-is-a-deal-breaker), for example.

RussTedrake, 2016-03-15:
It’s true. Previously, we had explicitly decided in drake that (named)
exceptions were ok. In fact I recently went through and changed the urdf parser
to use exceptions instead of passing back boolean success before starting the
sdf parser. We need to agree to either add this as an exception (pun feels
unavoidable) to the style guide or go back.
One very tangible motivation for exceptions was that it provides more useful
error messages that are handled relatively uniformly for the higher level
languages (like matlab, python), and even to the build servers.

liangfok, 2016-03-15:
I would be supportive of making an exception to the Google Style Guide to allow exceptions since we're not working with a large code base that is exception free.

PR #2584 (2016-06-21): Fixes Corner Case Where URDF Transmission Is for Non-Existent Joint.
Resolves #1864

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2584)
closed 2016-06-21 by liangfok