RobotLocomotion/drake #1867: Break links rather than joints for systems with topological loops

sherm1, 2016-03-15:
Often the best way to deal with loops in an internal-coordinate multibody system is to split within a *link* rather than at a *joint*. This has the substantial advantage that all joints are then treated uniformly and angular coordinates can wrap past 2π without introducing extra state variables or hacks. It requires sawing a link into two half-mass bodies and gluing them back together with a weld (fixed) constraint (6 equations). 

@RussTedrake adds: That would indeed be cleaner. The current implementation is limited in that we cannot easily add an actuator directly to the “loop joint”, nor friction, etc. And we’re already adding 6 constraints for a revolute joint anyhow (should be 5, but was cleaner to implement as 6). It would be a definite improvement!

(Discussion moved from #1714 with @liangfok)

sherm1, 2016-03-15:
A few notes:
- A ball joint is a reasonable exception; it is essentially the same modeled either way.
- Simbody has utility class `MultibodyGraphMaker` that calculates a good tree+constraints model for a given links+joints inputs. You can read about it [here](https://simtk.org/api_docs/simbody/latest/classSimTK_1_1MultibodyGraphMaker.html#details). (Simbody uses dynamicists' "body" as a synonym for roboticists' "link".)
- `MultibodyGraphMaker` is independent of Simbody and easily extracted if we want a similar capability for Drake. Gazebo uses it to make internal coordinate models from sdf input files.
- Assembly of weld constraints over large separations has to be done carefully; they can be inclined to assemble into an anti-parallel configuration.
- There is an efficiency argument to make for loop joints since they usually reduce both the number of differential and the number of algebraic equations. Other than for loop ball joints I have never seen this worth doing in practice. You wouldn't believe the agony we went through to add loop joints to SD/FAST -- Dan Rosenthal thought up the "break the bodies" idea afterwards causing a lot of face-palming.

liangfok, 2016-10-05:
Reading [this page](https://simtk.org/api_docs/simbody/latest/classSimTK_1_1MultibodyGraphMaker.html#details), I found the following two quotes:

 > A joint that would form a loop is marked as such, and then its child body is split to form a new "slave" body that can be mobilized by the joint. 

and:

> The algorithm normally decides which joints are the loop-breakers, however you can specify in the input that certain joints must be loop joints.

I was just wondering whether there is any bias regarding _which_ body to split and weld back together again using constraints to form a loop. I believe certain links like those based on primitive shapes will be easier to split than others that are, for example, based on meshes. If this is true, the algorithm should be biased towards splitting "easier to split" bodies.

sherm1, 2016-10-05:
There is a bias (actually a necessity) to avoid creating massless or inertialess terminal (leaf) bodies in the tree. So a massless body won't get chosen for a split. Otherwise, all bodies are equally easy to split; the slave body doesn't have to inherit any of the geometry, just half the mass properties. So a slave is actually a naked frame with mass properties attached; then that frame is welded to the real body.

liangfok, 2016-10-05:
I see. So another bias is how difficult it is to split the spatial inertia of a single body into two bodies. Is this trivial to do without specialized CAD software?

sherm1, 2016-10-05:
Splitting the mass properties couldn't be easier! If the original body's mass properties were mass,com,inertia=[m,c,I], the master and slave bodies each get mass properties [m/2,c,I/2]. No CAD software needed! In cases where n slaves are required, the denominator is (n+1).

BTW, the reason this isn't more obvious is that we treat the inertia strangely by convention. The mass properties are actually a *geometric* distribution of "mass moments" that are scaled by the mass; the COM always appears in the equations of motion multiplied by mass. If the inertia is represented as a "unit inertia" (@mitiguy calls that a "gyration matrix" G), then a body's spatial mass properties can be written as `m*[1,c,G]` where the mass has been factored out. Splitting now just means apportioning m/2 to each body; the mass moments don't change.

liangfok, 2016-10-05:
Oh wow, it's wonderful that `I` can be simply divided by two. What about the center of mass (`com`)? Why doesn't it change when a body is split in two? Intuitively, I would think that the new center of masses of each half-body, would be somewhere inside of each of the two half bodies.

sherm1, 2016-10-05:
> Why doesn't [the com] change when a body is split in two? 

See my comment above. None of the geometric mass moments change, just the mass (same as if we reduced the density by changing materials). The com is purely geometric (it is the first mass moment). A unit inertia (or gyration matrix) is the purely-geometric second mass moment. Unfortunately by convention we mass-scale it rather than separating the geometry from the mass as is done for com.

liangfok, 2016-10-05:
I see. I was interpreting "geometry" to mean "visual geometry". Got it, thanks!

tri-ltyyu, 2019-03-22:
Still relevant for when loops is supported in MBP (targeting Q2/Q3 2019).

RussTedrake, 2021-04-17:
Not sure if this issue still adds anything beyond #9566 and the [bushing element](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_linear_bushing_roll_pitch_yaw.html) ?

sherm1, 2021-04-18:
The implementation discussion above is still valuable, but works fine with the issue closed. I referenced it in #9566.

closed 2021-04-18 by sherm1