RobotLocomotion/drake #1868: CLion debug session "times out"

amcastro-tri, 2016-03-15:
I decided to give Clion another shot after adding more memory to my workstation. I followed the instructions [here](https://github.com/tkoolen/drake/wiki/CLion-setup-(experimental)) to set it up.

I am able to compile and run from within CLion. However when trying to debug (Shift+F9) the debugging panel opens up but however CLion stays there waiting for gdb. After a while (like about 2 minutes or so) I get a little bubble with the message "command timed out".

A google search on this revealed that apparently this is a know issue for large projects that CLion has been having even for the latest releases.

I tried their [Early Access Program](https://confluence.jetbrains.com/display/CLION/Early+Access+Program) (build # 145.257) version and this ones does start the debugger, but after a long wait of about 3 minutes!

Any word on how to get this running properly for Drake?

RussTedrake, 2016-03-15:
@tkoolen is our expert.

tkoolen, 2016-03-15:
Unfortunately I don't have any experience with debugging using gdb in CLion; I use lldb myself (on OSX).

Browsing CLion's issues, I found:
* https://youtrack.jetbrains.com/issue/CPP-4952. This one was resolved; it was due to using tcsh instead of bash; might want to check it out if you're using tcsh.
* https://youtrack.jetbrains.com/issue/CPP-3311. Error message is different, but could be related. Suggests that pointing CLion to a newer version of gdb may solve the problem.
* https://youtrack.jetbrains.com/issue/CPP-5508. Unresolved

amcastro-tri, 2016-03-16:
Anybody using CLion in Ubuntu with gdb?

sherm1, 2016-03-16:
> Anybody using CLion in Ubuntu with gdb?

@chrisdembia, are you still using CLion? If so do you have any advice for Alejandro (@amcastro-tri)?

chrisdembia, 2016-03-16:
I am still using CLion, but on OSX with lldb, like @tkoolen. I'm eager for version 1.5 so that I can choose the location of the build directory. Good luck Alejandro.

amcastro-tri, 2016-03-16:
I am considering to create ("generate" in cmake's terms) and IDE's project files with cmake option -G. I was considering to test this with Eclipse. Any inputs on that? is anybody else using another IDE?

chrisdembia, 2016-03-16:
Eclipse works with the UNIX Makefiles generate, I believe. And I've heard that Eclipse with C++ is pretty slow and doesn't handle big projects well. On Ubuntu, CLion is the best option, but you could also try Code::Blocks.

tkoolen, 2016-03-16:
I was frustrated by the lack of development being done on the Eclipse CDT. It also didn't handle templates well last I tried it.

RussTedrake, 2016-05-01:
i believe this discussion has runs its course, so will close it.

closed 2016-05-01 by RussTedrake

emmjaykay, 2018-02-08:
Still an issue with me on 2017.3. Ubuntu 16

amcastro-tri, 2018-02-09:
I believe @SeanCurtis-TRI debugs under CLion. I actually got used to the workflow of debugging with `gdb` and I use CLion for its awesome navigation features for C++ projects

aljohnston112, 2023-12-19:
I am getting timeouts. Makes debugging impossible after one evaluation.

CLion 2023.3.1
Build #CL-233.11799.287, built on December 11, 2023
Runtime version: 17.0.9+7-b1087.7 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 32768M
Cores: 12
Registry:
  debugger.attach.dialog.enabled=true
  run.processes.with.pty=TRUE
  ide.experimental.ui=true

sherm1, 2023-12-19:
You can adjust the timeout: https://www.jetbrains.com/help/clion/configuring-debugger-options.html#gdb-startup