RobotLocomotion/drake #1870: add a CMake target for building doxygen docs

sherm1, 2016-03-16:
It would be nice to be able to test doxygen along with code during development. Having CMake produce a "doxygen" target that could be built with `make doxygen` or from within an IDE would be useful and would hopefully encourage testing of doxygen output, which can be full of surprises.

liangfok, 2016-04-03:
I just noticed that [drake/doc/doxygen.h](https://github.com/RobotLocomotion/drake/blob/master/drake/doc/doxygen.h#L37) contains the following text:

```
If you want to make your own doxygen locally, run <b><i>make doxygen</i></b> in
the root drake directory. 
```
This implies that at some point in the past there _was_ a doxygen build target. I guess it must have been removed. 

I'm also curious how the "doxygen.h" file is used since I can't find it being referenced anywhere in Drake's code base.

sherm1, 2016-04-03:
`make doxygen` still works, Liang. Looks to have been hand written [here](https://github.com/RobotLocomotion/drake/blob/master/drake/Makefile#L58) in `drake/Makefile`; I don't think CMake knows about it.

liangfok, 2016-04-03:
Oh sorry, I now see you're wishing for a `CMake` target, not just a `make` target. It make sense to me to elevate the entire build API to the CMake level.

I just tried to build the doxygen content using the following command but it only generated a skeletal site with no classes or methods:

```
$ cd [drake distro]/drake
$ make doxygen
```
I tried this on an OS X machine only.

sherm1, 2016-04-03:
> I'm also curious how the "doxygen.h" file is used since I can't find it being referenced anywhere in Drake's code base.

That's a fake header file to hold doxygen's `@mainpage` stuff for Drake to generate the page that comes up first when you look at the generated documentation. Since it is named like a header, doxygen sees it while rummaging through all of Drake's source code when it visits the `doc` directory; it then extracts doxygen comments from it. You can see the result in Drake's main doxygen page [here](http://drake002.csail.mit.edu/drake/doxygen_cpp/). There is also a fake `doxygen.m` in that same directory for use when generating Matlab doxygen.

sherm1, 2016-04-03:
> I just tried to build the doxygen content using the following command but it only generated a skeletal site with no classes or methods

Doing the same thing on Windows produced complete doxygen docs. It ends up in `drake/doc/DoxygenCPP/html`. Is that where you're looking?

liangfok, 2016-04-03:
Yes, I am looking in `drake/doc/DoxygenCPP/html` and the generated website is essentially blank:

[image]

I also tried to execute the command directly as shown below, but got similar results:

```
$ cd [drake distro]/drake
$ doxygen doc/doxyfile_cpp
```

Until I can replicate this problem on more machines (haven't tried yet), I'm going to assume there's something wrong with my local machine and not escalate this into its own issue.

Here's the version of doxygen I'm using:

```
$ doxygen -v
1.8.11
```

sherm1, 2016-04-03:
I'm also using doxygen 1.8.11. If you didn't already, you might check the timestamp on index.html there to make sure it was actually generated by the doxygen run you just did.

sherm1, 2016-04-03:
A thought: did you run CMake at the top level yet? It may be doing some configuring of the doxygen files.

RussTedrake, 2016-04-03:
I can't think of any doxygen configuring happening in cmake.  All of the other advice here looks spot on.

PR #2128 (2016-04-21): Add CMake target for build documentation
Closes #1870.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2128)
commit b748f65b (2016-04-21): Add documentation CMake target

Closes #1870.
PR #2143 (2016-05-05): Add CMake target for build documentation (again)
Originally reviewed in #2128. Reverted in #2140. Closes #1811. Closes #1870.

Includes nasty hack from http://stackoverflow.com/questions/27979803/external-relative-link-in-sphinx-toctree-directive, unfortunately.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2143)
commit 650096ee (2016-05-05): Add documentation CMake target

Closes #1870.
closed 2016-05-05 by jamiesnape