RobotLocomotion/drake #1879: add solver options for linear systems of equations in the OptimizationProblem class

RussTedrake, 2016-03-17:
as pointed out by @edrumwri in #1874.  we should implement those options in particular.

sammy-tri, 2016-03-17:
For LCP solvers, some potential options include:

* Max number of pivots.  @edrumwri notes: The vast majority of problems are solvable within 4/3*N pivots (if memory serves), and too many pivots causes rounding error to accumulate, potentially causing the basis to become degenerate. Such degeneracy is also why I have avoided rank-1 updates to the matrix factorization.
 * Fast pivoting vs. Lemke
 * Regularization enabled / disabled (or should we always be using the regularized version?)

For other systems of equations, I think we'll need to see what NLopt has for relevant parameters which we want to encapsulate.  Another issue which I'll need to think about is what to do if we have a set of parameters for non-LCP problems where some solvers may ignore parameters which others use, and how to avoid misleading the user in this case.

RussTedrake, 2016-03-17:
The matlab NonlinearProgram class allows people to set snopt parameters.  (We need that in c++, too).  It might be a useful example.

sammy-tri, 2016-03-17:
Thanks, I'll start there.

edrumwri, 2016-03-18:
You'll always want to use the regularized version of Lemke- if no regularization is necessary, the solve time is the same as the non-regularized version.

For fast pivoting, it depends on the LCP matrix class. If the matrix is a P0-matrix (as occurs with frictionless problems, no-slip problems, complementarity-free problems, and projected constraint stabilization), then use the regularized version. If trying to apply the fast solver to complementarity-based problems with Coulomb friction, I _might_ call the un-regularized fast pivoting solver first- to see whether it can solve the problem quickly- and immediately move on to regularized Lemke if it fails.

sammy-tri, 2016-08-11:
Closing since #2238 allowed for generically setting solver options.  The MobyLCP content here was mostly speculative, and I think can be best addressed once MobyLCP has users who can identify their needs.

closed 2016-08-11 by sammy-tri