RobotLocomotion/drake #1880: LCMSystem APIs are too diffucult to understand and use

jwnimmer-tri, 2016-03-17:
A few of us from TRI were new to Drake a few weeks ago, and trying to understand how the Prius example worked.  One of the confusing aspects was how LCM messages were handled within Drake.

We feel that the design has substantial problems.  Here are our working notes:
* We tried to understand the interaction between LCMSystem.h, simulateLCM.cpp, and the LCM codgenned classes.
* The interaction depends on a complex mix of templating, ADL, operator= overloading, and header inclusion to work.
* The crucial moving piece is the call to decode() in LCMSystem.h. This _does not_ call the decode() function in that header. Rather via ADL it calls the global namespace decode() function defined in the compilation unit that included the header.
 * The crucial thing to understanding this code is that the template typename "Vector" refers not to any sort of vector but to the class that must be converted into a vector.  (The fact that Vector is a Concept is underdocumented / uncited.)
 * Yes, this is relying on overloading caused by global namespace collisions from the including file ​to work.
* The LCM message class (in this case DrivingCommand::LCMMessageType ?) is decoded via an overloaded assignment operator in an intermediate class (in this case "DrivingCommand").
* The intermediate class provides a toEigen() method that allows it to function as an input vector.
* Deep template magic here. The following examples are typical:
```
 template<template<typename> class Vector, typename Enable = void>
 template<template<typename> class Vector>
 class LCMOutputSystem<Vector, typename
   std::enable_if<!std::is_void<typename
   Vector<double>::LCMMessageType>::value>::type> {
```
* By the time an LCM message reaches the actual driving system it appears to have been converted to and from Eigen two or three times.
* It is our opinion that none of this code is remotely close to being readable, inspectable, or even usable in an important system.

My intent with this issue is to confirm the consensus that the design is a problem, and offer a forum for the several people who care about LCM integration to work towards a different implementation.

liangfok, 2016-03-17:
Another issue is there's no single point through which Drake interacts with LCM. In addition to the aforementioned LCMSystem.h, simulateLCM.cpp, and the LCM codgenned classes, another place where Drake interacts with LCM is [BotVisualizer.h](https://github.com/RobotLocomotion/drake/blob/82ecb839bd81e7d8f473f73ea0dc4d1a2b7dc888/drake/systems/plants/BotVisualizer.h). It would be nice to consolidate all Drake / LCM interactions into a single focal point so that other transport layers like ROS can be swapped without modifying numerous points in Drake's code.

jwnimmer-tri, 2016-03-17:
@liangfok: Thanks, that's another good example to consider.

In your BotVisualizer.h example, I'd certainly support replacing ```std::shared_ptr<lcm::LCM> lcm``` from that class with a recipient-agnostic signal; however, I'll posit that the ```mutable drake::lcmt_viewer_draw draw_msg``` should remain, as the signal argument.  At some point, you have to choose a canonical data repsentation, and LCM seems good enough for that.  When someone wants to integrate BotVisualizer with ROS, their code can instantiate a BotVisualizer and then connect its signal to code that transforms the LCM struct into the desired ROS message.

liangfok, 2016-03-17:
@jwnimmer-tri, yes I would agree to keep the LCM message type as the canonical data representation used internally within Drake, and to translate to messages used by other middleware systems at the periphery.

david-german-tri, 2016-03-17:
+1 for a recipient-agnostic, mock-friendly messaging interface.  I don't entirely agree that LCM is a good enough data representation, but we can gradually reinvent protocol buffers to fill in its deficiencies, as the needs arise.

jwnimmer-tri, 2016-03-17:
> I don't entirely agree that LCM is a good enough data representation

"Good enough" in terms of convenient support for lots of existing research, maybe?  I agree there are vastly superior alternatives.  I'd +1 a different issue to use a better library than LCM; my point to Liang was that I just don't want the canonical form to be ROS.

liangfok, 2016-03-17:
@david-german-tri are you suggesting the use of Protocol Buffers as the canonical data representation used within Drake?

david-german-tri, 2016-03-17:
@liangfok Yes, despite the switching costs.  I agree with @jwnimmer-tri that it's a separate issue, so I've opened #1881.

sherm1, 2016-03-17:
>  I would agree to keep the LCM message type as the canonical data representation used internally within Drake

Naive question: would it be better to use a transport-neutral rep like protobufs as Drake's native data structure?

sherm1, 2016-03-17:
> Naive question: would it be better to use a transport-neutral rep like protobufs as Drake's native data structure?

Sorry, late to party here in laid-back CA. I see #1881 now.

rpoyner-tri, 2016-03-17:
I got tired of writing Drake LCMVector concept code before I even finished my first wrapper:

<script src="https://gist.github.com/rpoyner-tri/a2de564a974629135edf.js"></script>

david-german-tri, 2016-05-19:
I think that System2 will make a much cleaner LCM publisher possible.  In the meantime, I'm going to close this issue because I don't think it leads to a specific item of work; there's no percentage in fixing up the System1 publishers at this point.

closed 2016-05-19 by david-german-tri