RobotLocomotion/drake #1881: LCM is not a sufficiently featureful internal data representation

david-german-tri, 2016-03-17:
As discussed in #1880, we would like to have one canonical data representation within Drake, that can be converted to many wire formats and sent over many transport protocols at the process boundary.  Currently, LCM is the de-facto standard for that representation, but it has some painful limitations:

- No reflection (and therefore no general, fine-grained comparator)
- No human-readable serialization
- No service definitions
- Manual memory management

Switching away from LCM would be burdensome, but it might be less burdensome than coping with all of the above indefinitely.  We should consider [protocol buffers](https://developers.google.com/protocol-buffers/), a robust, open-source data representation from Google.

patmarion, 2016-03-17:
Just a comment, in many of our openhumanoids projects, we try not to use lcm messages as internal data representations.  You may need to refactor your internal representation but you won't that to affect the message definition that you expose to external processes.

I agree with your assessment of the limitations of LCM.  I frequently use LCM from Python where there is reflection.  I use the lcm-spy (bot-spy) tool to inspect LCM traffic, so that satisfies my needs for human readable display.  Can you clarify the point about manual memory management?  The LCM c++ interface allows you to use stack allocation and stl vectors/strings, does it not?  One of my primary complaints about LCM is lack of support for versioning and backward compatibility with old message definitions.

My past experience with protocol buffers has been positive.  I don't have experience designing libraries or applications with protobuf for internal data representation, but I have worked third party libraries that do.

david-german-tri, 2016-03-17:
Revised the original post to clarify about memory management.

rpoyner-tri, 2016-03-17:
I got tired of writing Drake LCMVector concept code before I even finished my first wrapper:

https://gist.github.com/rpoyner-tri/a2de564a974629135edf

I believe a reflection-capable serialization layer may be of some help here.

jpieper-tri, 2016-03-18:
One use case that protobuf doesn't support particularly well is one that LCM (and other robotics systems I've seen), do particularly well, which is self describing data logs and live system introspection with generic tools.  

A common pattern is to include the schema in logged data, and then have tools which read that schema and data simultaneously to display it.  Similarly for general purpose live visualization tools.

This isn't impossible with protobuf, but it is challenging using the default upstream tooling, which for instance provides no pure-python way to at runtime load a schema and operate on it.  All you can do is compile the schema to a python file using the C++ compiled executable, then import that.

Another random gripe, is that the upstream protobuf python code still isn't python3 compatible.

traversaro, 2016-03-19:
It may be worth mentioning that Protobuf has also some well known performance problems, that had lead people to develop lightweight alternatives such as [FlatBuffers](https://google.github.io/flatbuffers/) or [https://capnproto.org/](Cap'n Proto).

[This discussion](https://bitbucket.org/ignitionrobotics/ign-transport/issues/23/consider-flatbuffers) on ign-transport issue tracker has some interesting pointer on this.

liangfok, 2016-03-19:
The pros/cons of all these data marshaling / de-marshaling libraries are fascinating and indicate there is no one-size-fit-all solution. The desire for features like reflection, human-readable serialization, and automatic handling of variable-sized fields come at the cost of computational resources and latency.

Going back to the title of this issue, I'd like to clarify what "internal data representation" means. Does it mean the data structures that are passed between objects or threads within a single process? Or, does it refer to the data structures that are passed between processes?

Within a single process, we should probably stick with C++ classes since that will maximize expressiveness both in terms of structuring the data and providing methods that operate on or access the data. This is already being done, for example, by [RigidBodyTree](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTree.h) and [KinematicsCache](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/KinematicsCache.h).

For inter-process communication, we'll probably need to work with several different data marshaling / de-marshaling mechanisms depending on the needs of the specific context. We should push the selection of a particular data marshaling / de-marshaling mechanisms as far out in the periphery as possible. Between the controller / simulator process and visualizer process, this is already being done by [BotVisualizer](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/BotVisualizer.h). The problem is BotVisualizer is currently hard-coded to only support LCM. How about let's make it a generic abstract class that is not specific to a particular data marshaling / de-marshaling mechanism, and allow sub-classes to specialize it to LCM, Protocol Buffers, Flat Buffers, raw structs, etc?

david-german-tri, 2016-05-19:
I think C++ classes are the de-facto consensus.

closed 2016-05-19 by david-german-tri

kannode, 2018-02-22:
Was the consensus to use LCM or Protocol Buffers. I am looking to your what drake converged on. We here, use pybind11 and RTTR for reflection, http://www.rttr.org/doc/master/classes.html to provide reflection and want to use LCM to serialize data into files and log events. RTTR might be useful for the reflection capabilities you see in drake.

EricCousineau-TRI, 2018-02-22:
The consensus here was to use basic C++ types for internal representations, and only use LCM / protobuf for message passing (and possibly some config files).

My conjecture is that we *may* consider something like `Boost.Serial` (or its C++11 cousin, `cereal`) for compile-time serialization exposure of non-numeric types to stay consistent with using basic C++ types; we will definitely keep an eye out for things like RTTR for run-time reflection mechanisms - thanks for pointing that out!
We are using `pybind11` as well, but solely for Python bindings. Can I ask which project y'all are employing `RTTR` + `pybind11` for?

kanaero, 2018-02-23:
This is a simulation tool used internally within our group (nodein.com) for Visual Odometry and SLAM algorithm development. We originally adapted binder to generate python bindings but recently moved to just writing a python script that would generate both the pybind11 and rttr code. My goal is to mark certain fields in our python script that indicate which fields should be serialized and automatically generated the POD datatypes in LCM or other tools to serialize directly.