RobotLocomotion/drake #1882: Changelog to document API and library changes

wxmerkt, 2016-03-17:
Great to see a lot of drake development activity and momentum. I'm still following every comment, issue, and PR, but wonder if it would be possible to establish a CHANGELOG document that would have a one-line description with a PR number reference that is added to the changelog file with each PR [1] that changes underlying functionality, API, ABI, or pkg-config/libraries, required header includes, etc., e.g.
- ``PR #1861 C++ API .simulate() now uses explicit trapezoid rule (RK2) instead of explicit Euler integrator (RK1) for better stability.``

It's rather challenging as a external/library user to otherwise follow what has changed since the last time that one updated drake without having to go through an entire diff log or fixing compile errors as they appear.

Looking forward to hearing your thoughts and comments,
Wolfgang

[1] Without versioning and with the difference in code review/discussion times for different PRs, dates might not be the best, so PR numbers seem to be a good compromise.

sherm1, 2016-03-17:
Great idea -- root level `CHANGELOG.md` would be good for a GitHub repo.

RussTedrake, 2016-03-17:
Agreed

closed 2016-04-02 by liangfok

liangfok, 2016-04-04:
@wxmerkt: Should the change log be sorted in the order the PRs were merged into the master branch or in the order in which they were created?

wxmerkt, 2016-04-05:
Good question @liangfok. Personally, I'd say in the order in which they were merged since some code review discussions for fundamental feature/algorithm changes take considerably longer than smaller changes. To avoid not spotting omissions, one would need to check the files-changed for the CHANGELOG.md file before hitting merge. Very happy to hear the input or best practices of more experienced OS leaders on this.

liangfok, 2016-04-07:
Thanks for the advice @wxmerkt. See https://github.com/RobotLocomotion/drake/pull/2014 for @jwnimmer-tri's updated version.