RobotLocomotion/drake #1887: Linker Complains of Duplicate Symbols from Inclusion of drake/util/testUtil.h

liangfok, 2016-03-18:
I want to use [valuecheckMatrix(...)](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/util/testUtil.h#L79) in multiple compilation units, specifically [RigidBody](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/systems/plants/RigidBody.cpp) and [RigidBodyTree](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/systems/plants/RigidBodyTree.cpp). Unfortunately, including testUtil.h in both .cpp files resulted in the following linker error:

```
Linking CXX shared library ../../lib/libdrakeRBM.so
CMakeFiles/drakeRBM.dir/RigidBody.cpp.o: In function `Eigen::DenseBase<Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Block<Eigen::Matrix<double, 6, 6, 0, 6, 6> const, 3, 3, false> const, Eigen::Transpose<Eigen::Block<Eigen::Matrix<double, 6, 6, 0, 6, 6> const, 3, 3, false> > const> >::isZero(double const&) const':
/home/liang/dev/drake/build/include/eigen3/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1263: multiple definition of `valuecheck(double, double, double)'
CMakeFiles/drakeRBM.dir/RigidBodyTree.cpp.o:/home/liang/dev/drake/drake/../drake/util/testUtil.h:116: first defined here
```
The problem is because [valuecheck(...)](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/util/testUtil.h#L116) is implemented in the same header file resulting in the implementation being included twice. To work around this problem, I modified the method to be `inline`. Here is the commit: https://github.com/liangfok/drake/commit/58d3bb12827f84da2d9b1dbc4367d1e38ffdca7f.

Is this a valid fix?

RussTedrake, 2016-03-18:
Does the google test suite have a suitable replacement for the valuecheck 
logic? Perhaps we will need something like this to turn an eigen call into 
a boolean pass/fail?

sammy-tri, 2016-03-18:
A couple of thoughts...

For fixing the duplicate implementation of valuecheck, my first thought would be to move it into a .cpp file and only leave the forward declaration in the header.  If it's performance critical, inline is probably fine.  (I'd say "static inline" if this were pure C). 

My first thought on a way to handle this issue (which I think is more in the spirit of #1743) would be to create a googletest fixture (https://github.com/google/googletest/blob/master/googletest/docs/Primer.md#test-fixtures-using-the-same-data-configuration-for-multiple-tests ) which has functions in the fixture for comparing matrices...  wrapping the valuecheck family of methods in a class would presumably be a working defence against linker errors, since class member functions are defined in headers all the time.

liangfok, 2016-03-19:
Looking at [Google Test's binary comparison operators](https://github.com/google/googletest/blob/master/googletest/docs/Primer.md#binary-comparison), they do not support equality tolerance values. Thus, Drake will need to maintain these specialized `valueCheck()` and `valueCheckMatrix()` methods that accept tolerance values and ideally push them upstream into Eigen itself.

TODO: Verify that Eigen's [isApprox()](http://eigen.tuxfamily.org/dox/classEigen_1_1DenseBase.html#a158c2184951e6e415c2e9b98db8e8966) is not a substitute for Drake's `valueCheckMatrix()` particularly in terms of handling infinity and NaN values.

Assuming Eigen's `isApprox()` is insufficient, I vote to move these valueCheck methods into a class that can then be instantiated and used across multiple Google Test fixtures.

sherm1, 2016-03-19:
Google Test does have a set of [floating point comparison macros](https://github.com/google/googletest/blob/master/googletest/docs/AdvancedGuide.md#floating-point-macros) including `ASSERT_NEAR()` and `EXPECT_NEAR()` that take tolerances. I'm not sure how useful these would be when comparing matrices though.

liangfok, 2016-03-19:
Thanks @sherm1 ! Those Google Test floating point comparison macros do indeed seem to be a sufficient replacement for Drake's [valuecheck()](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/util/testUtil.h#L116) method. Agreed it is not sufficient for comparing Eigen matrices.

Surprisingly, the tolerance value to use seems to be non-trivial based on [this blog post](http://realtimecollisiondetection.net/blog/?p=89).

sherm1, 2016-03-19:
> Surprisingly, the tolerance value to use seems to be non-trivial based on this blog post.

Not surprising to me -- Erin Catto (mentioned in that post) worked for me at two companies! Although the abstol/reltol distinction is commonly made in numerical software, I don't use it because I have found in practice that it is extremely rare for anyone to know how to set them both correctly. Instead, the relative tolerance is the important part and is easy enough to understand because, for example, 0.01 means "within 1%". That has to be relaxed for values near zero; the way Christer Erickson does it in that post is good and it can be improved further if you have some scaling information available. IMO asking for abstol is asking for trouble.

liangfok, 2016-03-19:
Wow!  What a coincidence!  

I'm now wondering whether the reason why the equivalency test between my URDF and SDF models of the Prius fails is due to the use of an absolute tolerance value in [valueCheckMatrix(...)](https://github.com/RobotLocomotion/drake/blob/fa5cbb18d703fe133ede5ea01f7a0a4953bc9c21/drake/util/testUtil.h#L79), which uses Eigen's [isZero(...)](http://eigen.tuxfamily.org/dox/classEigen_1_1DenseBase.html#a6dc0eb069536157ead1d46c660b9622d). This is because the mismatch occurs when huge numbers on the order of 1e5 appear in the return value of `RigidBodySystem::dynamics(...)`.

sherm1, 2016-03-19:
Yeah, even if everything was expected to match to machine precision (~2e-16) two numbers of magnitude around 1e5 calculated perfectly would differ in the 11th decimal place.

liangfok, 2016-03-20:
Thanks! It makes perfect sense. I will be more careful selecting absolute / relative tolerance values when comparing floating point numbers in the future.

david-german-tri, 2016-03-22:
@liangfok points out that #1916 is closely related to this thread.  Ideally, we would have a [custom googlemock matcher](https://github.com/google/googletest/blob/master/googlemock/docs/CookBook.md#extending-google-mock) for Eigen matrix equality:

EXPECT_THAT(actual_matrix, EigenMatrixIsApproximatelyEqual(expected_matrix));

I've never written a matcher for a template class before, but I think it's possible (in the long form, not with the convenience macros).  

If the matcher doesn't pan out, I would propose simply moving the implementation of valuecheckMatrix into a .cc file as @sammy-tri originally suggested.  As a rule of thumb, if you want to initialize some resources a particular way and make them available to all the test cases, you need a test fixture.  If you merely want to share reentrant code across test cases, vanilla functions and libraries do the job.

closed 2016-04-11 by RussTedrake