RobotLocomotion/drake #1889: Installation Failure

banerjs, 2016-03-18:
I just tried to install Drake on a Windows 10 (build 1511) with Visual Studio 2013 x64. Everything works well until a consistent failure towards the end of the `make` step. Here are the last few lines of the make command:

```
...
...
...
  Done Building Project "<dir>\MATLAB\drake-distro\externals\yaml_cpp\pod-build\install.vcxproj" (default targets).
  
  Build succeeded.
      0 Warning(s)
      0 Error(s)
  
  Time Elapsed 00:00:06.78
  Completed 'yaml_cpp'
FinalizeBuildStatus:
  Deleting file "x64\Release\yaml_cpp\yaml_cpp.tlog\unsuccessfulbuild".
  Touching "x64\Release\yaml_cpp\yaml_cpp.tlog\yaml_cpp.lastbuildstate".
Done Building Project "<dir>\MATLAB\drake-distro\pod-build\yaml_cpp.vcxproj" (default targets).
Done Building Project "<dir>\MATLAB\drake-distro\pod-build\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.

"<dir>\MATLAB\drake-distro\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
"<dir>\MATLAB\drake-distro\pod-build\drake.vcxproj" (default target) (19) ->
(CustomBuild target) -> 
  C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targets(170,5): error MSB6006: "cmd.exe" exited with code 1. [<dir>\MATLAB\drake-distro\pod-build\drake.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:04:15.38
```

Any idea what's happening? Brief Google searches have revealed that I might need to upgrade to Visual Studio 2015, but I want to check with you guys before taking that drastic step

david-german-tri, 2016-05-19:
We only test on Visual Studio 2015.  I can't promise that upgrading will fix your problem, but it sounds like the logical next step to me.

http://drake.mit.edu/developers.html

Closing this issue for now.  Please feel free to reopen if you have additional evidence of a build problem in Drake on a supported platform.

closed 2016-05-19 by david-german-tri

liangfok, 2016-05-19:
Also note the updated Windows install instructions: 
http://drake.mit.edu/windows.html