RobotLocomotion/drake #1895: Integration of C++ Logging Framework with Drake

liangfok, 2016-03-20:
For debugging and diagnostic purposes, it would be nice to integrate a quality logging framework with Drake. Practically anything would be an improvement over `std::cout`, which is what I'm currently using. At a minimum, I'm thinking it should provide:

1. multiple verbosity levels (e.g., [here are ROS' verbosity levels](http://wiki.ros.org/Verbosity%20Levels))
2. timestamps
3. indicator of location from which message was generated (e.g., file name, line number, class name, method name, etc.) 
4. ability to print log messages to terminal and optionally save to a file or transmit over a network for saving on remote computer
5. low overhead for real-time operation and yet ensure probability of loosing a log message is near zero 
6. be compatible with Drake's open source license
7. match Drake's cross platform support (Linux, OSX, Win32, Win64)

It would be nice if the framework includes a strong set of tools for analyzing the resulting log files and mining / visualizing the data contained within them.

Some possible candidates:

| Framework | License | Notes |
|----------------|-------------|---------|
| [log4cxx](https://logging.apache.org/log4cxx/index.html) | Apache 2.0 |  used by ROS, see [here](http://wiki.ros.org/roscpp/Overview/Logging)|
| [log4cplus](https://sourceforge.net/projects/log4cplus/) | Apache 2.0 |  |
| [log4cpp](http://log4cpp.sourceforge.net/) | LGPLv2.1 | used by Orocos, see [here](http://www.orocos.org/wiki/rtt/examples-and-tutorials/using-real-time-logging) |
| [spdlog](https://github.com/gabime/spdlog) | MIT | |
| [Pantheios](http://www.pantheios.org/) | BSD | |
| [ACE](http://www.cs.wustl.edu/~schmidt/ACE.html) | ? | |
| [P7](http://baical.net/p7.html) | LGPLv3 | |

sherm1, 2016-03-20:
I like this idea. Of the ones you listed, I thought [spdlog](https://github.com/gabime/spdlog) looked most promising. I liked the emphasis on speed, especially the compile-time disabling of debug and trace logging so that those can be truly zero cost when not in use. Also, it is lightweight, header only, written last year in fresh C++11 style, and covered under a trouble-free license.

RussTedrake, 2016-03-20:
in all of our previous work, we've leaned on the lcm logging tools.  adding logging to a file meant adding a few lines (no more than what i see in spdlog) to publish a debugging lcm message.  then it goes in the logs, can be played back, etc.

i'm willing to concede that the lcm approach doesn't capture all use cases, and might not scale to very complex projects.  but it's worth discussing as (perhaps the easiest) alternative?

sherm1, 2016-03-20:
Is the LCM logger restricted to message logging or can it also be used to log arbitrary occurrences in a C++ program?

RussTedrake, 2016-03-20:
we can make a default logging type (e.g. with an enumeration so you can log different variable types) and then just populate the structure and call publish. we could obviously wrap that up behind a logging interface.
advantages would include playback and graphical visualization tools, and that the data would go into the same log files we’re using elsewhere.
disadvantages would include possible limits on data-rate/scale, and potential loss of precise timing info (not sure how that is handled by the other libs).

liangfok, 2016-03-21:
I'm currently leaning towards using [spdlog](https://github.com/gabime/spdlog) with [LCM](https://lcm-proj.github.io/) as a custom [sink](https://github.com/gabime/spdlog/wiki/4.-Sinks). Adding LCM as a sink should be as easy as extending [spdlog::sink](https://github.com/gabime/spdlog/blob/master/include/spdlog/sinks/sink.h) and implementing [log(...)](https://github.com/gabime/spdlog/blob/master/include/spdlog/sinks/sink.h#L19). The motivation for using spdlog + LCM rather than LCM directly is due to spdlog's logging-relevant API (e.g., log levels, log format, log macros, etc., details on [spdlog's wiki](https://github.com/gabime/spdlog/wiki)) and support for numerous types of sinks.

sherm1, 2016-03-21:
One thing to think about regarding spdlog (an instance of a general problem, see #1873) is this: since Drake is a library it will be incorporated into other programs (mains, GUIs, etc.). Suppose one of those programs was already using spdlog when they decided to incorporate Drake. Could the two loggers peacefully coexist? Would there be compile- or link-time conflicts? What if the two versions of spdlog were different, with incompatible APIs?

liangfok, 2016-03-21:
Good point. I suppose the same question applies to all of Drake's dependencies including LCM itself. What is the best way to answer this question other than to actually try to integrate Drake into another project that uses a different version of a particular dependency with an incompatible API or ABI? Can it be done by looking at the dependency's source code or analyzing the symbols in the compiled object files?

Maybe forking a dependency and putting all of its code inside Drake's namespace would allow linking but conflicts may still arrise if the dependency uses a shared resource like CPU, GPU, memory, network interface, etc.

jwnimmer-tri, 2016-03-24:
> I'm currently leaning towards using spdlog with LCM as a custom sink ...

This seems like a fair choice, and good experiment to try.

> ... since Drake is a library it will be incorporated into other programs (mains, GUIs, etc.) ...

The "drake way" would be to make the spdlog optional, right (under "make options")?  I believe that something like "#include <drake/logging.h>" with a fairly narrow API in the drake namespace would suffice (create a Logger, given a name; call debug()/info()/... on it; left-shift in the message contents).  If you've enabled spdlog in the build, you get the spdlog implementation.  If not, you get a no-op API stub that we write.  If you want something else, well then fork Drake and reimplement our logging.h concept.

Note that this only works if (1) we choose to use a very small API subset and (2) we spin up a CI variant that tests the spdlog-disabled configuration.

sherm1, 2016-03-24:
> The "drake way" would be to make the spdlog optional

It would also be possible to thinly wrap the spdlog API so that the Drake API exposes only logging symbols in the Drake namespace (or maybe `Drake::log::`), and macros `DRAKELOG_DEBUG()` or whatever where these would map exactly to the spdlog version we build with. This would have the advantage that calling programs could still use spdlog for logging their own information, with Drake's logging available but independent. The disadvantage (advantage?) would be that the logs would not be integrated, although that could be obtained if the calling program made use of Drake's logger instead of a separate spdlog.

Making logging optional would be awkward because the spdlog interface is mostly real methods; macros are used for TRACE and DEBUG so they can be compiled out. It would be nice to be able to use the logging API as is without macro-fying it.

jwnimmer-tri, 2016-03-24:
@sherm1 I'm pretty sure my proposal actually addresses those concerns, but perhaps I didn't explain it well.  I will write up a small code example to illustrate.

jwnimmer-tri, 2016-03-24:
Here is the rough outline of my proposal: [text-logging-sketch](https://github.com/RobotLocomotion/drake/compare/master...jwnimmer-tri:text-logging-sketch).  When HAVE_SPDLOG is turned off, the logging statements in client code will still compile, but will be dead-code eliminated by the compiler.

sherm1, 2016-03-24:
@jwnimmer-tri, thanks for the sketch. That looks nice. It has the advantage of allowing compile time elimination of all logging. That could be useful if only to check whether we are spending a lot of time in the logger. It would be important to verify that the compilers properly evaporate all the fake API code. [This line](https://github.com/RobotLocomotion/drake/compare/master...jwnimmer-tri:text-logging-sketch) makes me nervous -- if not eliminated I think it will allocate a new logger on the heap for every disabled log message, then delete it! In Debug that would always happen which would be disturbing though probably harmless. Also, it may still require some interface work for the spdlog macros to expand into the `Drake::logger` namespace when enabled; it would be nice to get spdlog modified to allow injection of an additional namespace -- I bet Drake wouldn't be the only use case for that.

jwnimmer-tri, 2016-03-25:
> [call to new to populate the shared_ptr] makes me nervous

Oops, that's my mistake in the drafting.  The stub should be returning a singleton logger, not a fresh one every time.

> Also, it may still require some interface work for the spdlog macros ...

My proposal would be to never use those macros within drake -- to only allow the call sites to look like the one example I changed in LQR.h by way of example.  I posit they they add only minimal value -- that even with spdlog enabled at compile time (but tracing disabled at runtime), the cost of the `if (enabled_) { ... }` checks that end up being run get lost in the noise.

sherm1, 2016-03-25:
> My proposal would be to never use those macros within drake

I would prefer that we use the spdlog interface unchanged -- I think it has been thought through very nicely and that will also allow us just to point to its documentation. In particular the TRACE and DEBUG macros are well-conceived IMO. Cases where the flag-check matters are rare, but I have seen them in practice and don't want to have to think about that whenever I want to add one of those.

(The most common scenario where these matter involves suppression of inlining that would otherwise get done if the code weren't there.)

jwnimmer-tri, 2016-03-25:
I had misremembered how many macros there were, and their nature.  Looking again now, I am fine with stubbing those two macros for use as well -- they are trivial for us to support.  I've updated my branch with that implementation (and changed LQR.h to use it instead).

sherm1, 2016-03-25:
There is still a lot of spdlog API not wrapped, much of which looks very useful (see example [here](https://github.com/gabime/spdlog/blob/master/README.md#usage-example)). I suppose we could stub more as needed but that seems like a pain and would require CI support as you mentioned to make sure we hadn't missed anything. Would it be sufficient just to wrap spdlog in `namespace Drake` to avoid namespace corruption and not try to make it optional? TRACE and DEBUG would be compiled out anyway.

jwnimmer-tri, 2016-03-25:
Which unstubbed features -- formatting?  I'd be game for adding more stubs for any spdlog APIs that Drake developers strongly want.  As much as I want to keep the stub API minimal, it's really not that hard to add new stub methods / macros.

However, in code like Drake, I think that if you're trying to use fancy formatting from a text logging API, you've probably already failed.  I posit that non-throwaway logging should be done in structured, self-describing binary form with tool support for offline analysis, plotting, etc.  It is not very difficult to do that once you have something like #1881 fixed.  Basically, it should be easy for every instance of every matrix, input, output vector, collision, etc. in a Drake execution to be dumped to disk in full trajectory form, and then plotted and explored interactively.  That's often what people use text logging for, badly.

For now, I want to get Drake to have _some_ text logging answer, because (1) thirdParty code we copy and edit in-tree wants to have such a thing (e.g., that code currently logs to std::cout and we need to redirect that somehow), and (2) people currently use printf debugging, and giving them text logging for now is better than commented-out std::couts.

Stepping back a bit, and addressing the line of thought that "this stubbing is all a bit complex / nasty for just being able to use a logger"... well, my proposal there would be to instead just directly and unconditionally use spdlog (without changing its namespace) in Drake, and to defer any obsessing about transitive linking conflicts in some future-Drake-as-library world.  It really won't be that hard to address that problem later once it actually happens.  You can even argue that using spdlog without re-namespacing it is a _feature_, because it means that code that integrates Drake along with other spdlog users will be able to centrally configure and manage its logging, without having to install special sinks to channel drake::spdlog::Sink into their own spdlog::Sink.

sherm1, 2016-03-25:
> Stepping back a bit, and addressing the line of thought that "this stubbing is all a bit complex / nasty for just being able to use a logger"... well, my proposal there would be to instead just directly and unconditionally use spdlog (without changing its namespace) in Drake

That sounds very reasonable. No rush to solve a problem we don't have!

jwnimmer-tri, 2016-04-13:
I'm not maintaining my example branch anymore.  Here's the outline I wrote up, as a git format-patch.

[0001-Add-spdlog-logging-sketch.patch.txt](https://github.com/RobotLocomotion/drake/files/218086/0001-Add-spdlog-logging-sketch.patch.txt)

jwnimmer-tri, 2016-07-26:
Tagging @ggould-tri here, as he as expressed interest.

jwnimmer-tri, 2016-07-26:
Notably, between these discussions four months ago and now, spdlog has removed the streaming API in gabime/spdlog#236, with the goal of an even more minimal API.  This should make it even easier to stub.

sherm1, 2016-07-27:
> This should make it even easier to stub.

I've forgotten why that's even a good idea. Why don't we just adopt spdlog as-is and use it?

ggould-tri, 2016-07-27:
(1) I am in the process of working this, and am assigning myself onto it, because I had to do a horrible thing in my code to get this effect.
(2) I believe the stub requirement is the general requirement that Drake should be able to exist without its dependencies being available, just as it does with solvers.

ggould-tri, 2016-07-28:
Fixed in #2954.

closed 2016-07-28 by ggould-tri