RobotLocomotion/drake #1897: Shouldn't use ALL_CAPS for constants

sherm1, 2016-03-20:
Because macros are very dangerous out-of-language loopholes, they are conventionally given horrifying names to make sure programmers notice them and treat them with appropriate trepidation. The Google style guide uses `MY_MACRO_NAME_THAT_SCARES_SMALL_CHILDREN` as a typical example. Long ago in C the only way to introduce constants into a program was to use macros, leaving them untyped and dangerous, so they inherited this naming convention. C++ fixed that problem by adding in-language constants, which are not dangerous and do not need to demand a programmer's attention the way macros do.

Google's discussion is [here](https://google.github.io/styleguide/cppguide.html#Constant_Names) and uses the convention `kThisIsAConstant`. Simbody's is [here](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md#constants) and uses the even milder convention `ThisIsAConstant`. I'd be happy with either of those but I would like to avoid `CRYING_WOLF` except for truly scary things.

RussTedrake, 2016-03-21:
We should follow the style guide unless there is a strong reason to make the exception.

Again, i don't know it well enough yet to police it in my code reviews.

liangfok, 2016-03-22:
I presume [header guards](https://google.github.io/styleguide/cppguide.html#The__define_Guard) are not classified as macros even though they, like macros, are defined using the `#define [macro name]` syntax?

david-german-tri, 2016-03-22:
:+1: for the original post - what @sherm1 suggests is both style-guide-compliant and a good idea.

@liangfok Header guards are definitely macros, but as you note they have special status in the style guide.

sherm1, 2016-03-22:
> I presume header guards are not classified as macros

They are definitely macros and should use `UGLY_CAP_NAMES_LIKE_THIS_H_`. (The final `_H_` is conventionally appended so they won't bump into any other macro names.)

sherm1, 2016-09-08:
We are gradually converting constants to the `kGoogleStyle` and have gotten rid of header guards altogether so I don't think we need to track with this issue any more.

closed 2016-09-08 by sherm1