RobotLocomotion/drake #1900: Windows build fails if 64 bit python is installed

MikeElkins-TRI, 2016-03-21:
If one follows the build directions on a machine with 64 bit python 2.7 installed, the build will eventually fail in the lcm directory, with unresolved glib externals (g_ptr_array_new is the first one mentioned, for people searching issue text).

This is due to the lcm-python wrapper being compiled (correctly) for 64 bits to match the python, but gtk (which includes glib) doesn't build, it unpacks a zip file of pre-compiled, 32 bit binaries, for reasons unknown to this author.

This could be fixed by not precompiling gtk, or including both 32 and 64 bit binaries, or not compiling lcm-python if the python is 64 bit, or by just adding documentation to the build instructions warning of the incompatibility, in approximate order of desirability.

billhoffman, 2016-03-21:
@jhoare I thought we addressed this already.

jhoare, 2016-03-21:
I think this is related but somewhat different, if I'm reading this correctly, it looks like 64 bit python is installed, but Mike is trying to build the drake as 32bit. Is this correct?

MikeElkins-TRI, 2016-03-21:
The bit-ness of drake doesn't matter, I've tried both.

billhoffman, 2016-03-21:
No that is not correct.  The problem is gtk is 32 bit even for 64 bit.  I thought that was fixed.

jhoare, 2016-03-21:
32bit gtk is downloaded even when you're building in 64bit mode? 

When building 64 bit are you setting CMAKE_FLAGS to -G "Visual Studio 14 2015 Win64"  like in: http://drake002.csail.mit.edu/drake/sphinx/windows.html

jhoare, 2016-03-21:
There used to be a bug that 32bit gtk was always installed but that should be fixed. 

The check is here: https://github.com/RobotLocomotion/gtk-pod/blob/master/CMakeLists.txt#L19

MikeElkins-TRI, 2016-03-21:
CMAKE_FLAGS: No I hadn't done that (as it is in a part labeled "If you have
Matlab 64bit"), but I had selected the 64 bit compiler when asked during
"make options", which I assume has the same effect.

Building on my machine with 64 bit python:
file="C:/Users/Elk/drake-distro/externals/gtk/pod-build/gtk3-prefix/src/gtk+-bundle_3.6.4-20130921_win32.zip

jhoare, 2016-03-21:
Actually selecting the 64 bit compiler during ``make options`` will not be the same since some pods only get the information that they are building for 64 bit via that environment variable. 

See here: https://github.com/RobotLocomotion/drake/pull/1843#issuecomment-194916484

Although I wonder if there as still a separate issue of having 64bit python and a 32bit build?

jhoare, 2016-03-21:
I was able to successfully build drake with a 64 bit python when I had the environment variable set. 

When  attempting to build 32bit, I got linker errors as you described. 

It looks like lcm's CMakeLists.txt should be updated to check that the architecture is equal to what version it is attempting to build. Right now it appears to only let it through if it is 64bit. 
https://github.com/RobotLocomotion/lcm-pod/blob/cmake/lcm/lcm-src/CMakeLists.txt#L62

MikeElkins-TRI, 2016-03-21:
Should the environment variable be set for all windows users, or only users
using Matlab64 bit OR python 64 bit?

MikeElkins-TRI, 2016-03-21:
I confirm that python64 build succeeds with CMAKE_FLAGS set to 64 bit.

Should we change the windows build instructions, or make it fail more
loudly in this situation?

jhoare, 2016-03-21:
RobotLocomotion/lcm-pod#18 and #1903 for the python checking in lcm.

billhoffman, 2016-03-21:
For now I guess we should update the build instructions.  I think there might be a way to pass the -G flag to the make pods by adding a make CMAKE_FLAGS="-G \"${CMAKE_GENERATOR}\" to the make call.

jhoare, 2016-03-22:
Agreed, #1909 can be used to track the requirement of CMAKE_FLAGS environment variable.

PR #1903 (2016-03-23): LCM tag to include RobotLocomotion/lcm-pod#18
Fixes #1900
closed 2016-03-23 by RussTedrake