RobotLocomotion/drake #1902: drake006 is failing because of the custom ctest

RussTedrake, 2016-03-21:
the generated CTestCustom.cmake looks like:
```
set(CTEST_CUSTOM_PRE_TEST "/usr/bin/perl /tmp/drake-5642/drake/cmake/add_matlab_
unit_tests.pl /tmp/drake-5642/drake")
```
which is generated from the command in CMakeLists.txt 
```
    file(WRITE ${CMAKE_BINARY_DIR}/CTestCustom.cmake
   	      "set(CTEST_CUSTOM_PRE_TEST \"${PERL_EXECUTABLE} ${PROJECT_SOURCE_D
IR}/cmake/add_matlab_unit_tests.pl ${PROJECT_SOURCE_DIR}\")\n")
```
calling ctest then fails because the quotes are in the wrong place (it say "program not found").  

after the cmake upgrade (++), this command is now failing.  probably a trivial fix for kitware.

RussTedrake, 2016-03-21:
@jamiesnape -- tagging you so you're in the loop, too.

jamiesnape, 2016-03-21:
Thanks. For reference, the CMake upgrade was from 2.8.8 to 2.8.11.

billhoffman, 2016-03-21:
I am not able to reproduce this.   I tried with 2.8.11. It seemed to work.  I put this into the CTestCustom.cmake file in CMake itself:

set(CTEST_CUSTOM_PRE_TEST "/usr/bin/perl /tmp/drake-642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake")

Then created a perl script that did this:
#!/bin/perl
print "$ARGV[0]\n";
print "$ARGV[1]\n";
print "hello\n";

It seems to work:

hoffman@hoffman-VirtualBox:~/My Builds/cmake-build$ ./bin/ctest -R COnly
Test project /home/hoffman/My Builds/cmake-build
/tmp/drake-5642/drake

hello
Guessing configuration NoConfig
    Start 37: COnly
1/1 Test #37: COnly ............................   Passed    0.27 sec

100% tests passed, 0 tests failed out of 1

RussTedrake, 2016-03-22:
i also confirmed (yesterday) that running the string inside the quotes at the command line worked just fine.

billhoffman, 2016-03-22:
Can you upload the file?  And also post the exact error message you are seeing?

RussTedrake, 2016-03-23:
the exact error is visible on any drake006 CI:

```
Ready to build!  pull-liangfok-feature/debug_sdf_cleanup-into-RobotLocomotion-master
-- CMAKE_INSTALL_PREFIX=/tmp/drake-1898-685/build
CMake Warning at cmake/mex.cmake:241 (message):
  Your MEX compiler flags contained '-ansi', but we've removed that flag for
  compatibility with C++11
Call Stack (most recent call first):
  CMakeLists.txt:120 (mex_setup)

-- Found matlab
-- Found bot2-lcmgl
-- Found eigen3
-- Found octomap
SWIG_EXECUTABLE: /tmp/drake-1898-685/build/bin/swig
Checking if Swig has Matlab support... yes
Found Swig with Matlab support
-- Writing addpath_drake.m and rmpath_drake.m to /tmp/drake-1898-685/build/matlab
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/drake-1898-685/drake/pod-build
   Site: drake006
   Build name: pull-liangfok-feature/debug_sdf_cleanup-into-RobotLocomotion-master
Create new tag: 20160323-1309 - Experimental
Configure project
   Each . represents 1024 bytes of output
    . Size of output: 0K
Build project
   Each symbol represents 1024 bytes of output.
   '!' represents an error and '*' a warning.
    ........ Size of output: 7K
   0 Compiler errors
   0 Compiler warnings
Test project /tmp/drake-1898-685/drake/pod-build
No such file or directory
Problem running command: /usr/bin/perl /tmp/drake-1898-685/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-1898-685/drake
Problem executing pre-test command(s).
Performing coverage
 Cannot find any coverage files. Ignoring Coverage request.
Submit files (using https)
   Using HTTP submit method
   Drop site:https://drake-cdash.csail.mit.edu/submit.php?project=Drake
   Uploaded: /tmp/drake-1898-685/drake/pod-build/Testing/20160323-1309/Build.xml
   Uploaded: /tmp/drake-1898-685/drake/pod-build/Testing/20160323-1309/Configure.xml
   Submission successful
Errors while running CTest
Unrecognized MATLAB option "wait".
                            < M A T L A B (R) >
                  Copyright 1984-2012 The MathWorks, Inc.
                    R2012b (8.0.0.783) 64-bit (glnxa64)
                              August 22, 2012

 
To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
 
{Error using xmlstringinput (line 43)
File pod-build/Testing/20160323-1309/Test.xml not found

Error in xmlread (line 77)
   fileName = xmlstringinput(fileName,true);

Error in postProcessCTest (line 12)
testfile = xmlread(fullfile(testpath,d.name,'/Test.xml'));
} 
>> 
                            < M A T L A B (R) >
                  Copyright 1984-2012 The MathWorks, Inc.
                    R2012b (8.0.0.783) 64-bit (glnxa64)
                              August 22, 2012

 
To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.
 
{Error using xmlstringinput (line 43)
File pod-build/Testing/20160323-1309/Test.xml not found

Error in xmlread (line 77)
   fileName = xmlstringinput(fileName,true);

Error in postProcessCTest (line 12)
testfile = xmlread(fullfile(testpath,d.name,'/Test.xml'));
}
```

billhoffman, 2016-03-24:
Not sure what to do next...  You should be able to test by changing into the build directory and running a single test:

cd /tmp/drake-1898-685/drake/pod-build
ctest -R sometest  --debug -VV

The -VV might shed some light on the issue as it will show the command that is about to be run.

RussTedrake, 2016-03-25:
It’s not a test that’s failing, it’s the pretest script. Running with `—debug -VV` confirms this:
``` ../../Source/CTest/cmCTestTestHandler.cxx:547 Test project /tmp/drake-1931-12489/drake/pod-build
../../Source/CTest/cmCTestTestHandler.cxx:1308 Run command: /usr/bin/perl\ /tmp/drake-1931-12489/drake/cmake/add_matlab_unit_tests.pl\ /tmp/drake-1931-12489/drake No such file or directory
../../Source/CTest/cmCTestTestHandler.cxx:1314 Problem running command: /usr/bin/perl /tmp/drake-1931-12489/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-1931-12489/drake
../../Source/CTest/cmCTestTestHandler.cxx:481 Problem executing pre-test command(s). ```
I was pretty confident that it was the quotes breaking things…

billhoffman, 2016-03-25:
OK, that is what I wanted to see.   /usr/bin/perl\ /tmp/drake...  Note the "\ ".   Have to figure out how/why that is happening.  When I have this: set(CTEST_CUSTOM_PRE_TEST "/usr/bin/perl /tmp/drake-642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake"), it runs the pre-test without escaping the spaces.  

When I run it I get this:

set(CTEST_CUSTOM_PRE_TEST "/usr/bin/perl /tmp/drake-642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake")

./bin/ctest --debug -VV -R COnly  | grep perl
/home/hoffman/My Builds/cmake/Source/cmCTest.cxx:2661   -- /usr/bin/perl /tmp/drake-5642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake
/home/hoffman/My Builds/cmake/Source/cmCTest.cxx:2661   -- /usr/bin/perl /tmp/drake-5642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake
/home/hoffman/My Builds/cmake/Source/CTest/cmCTestTestHandler.cxx:1333 Run command: /usr/bin/perl /tmp/drake-5642/drake/cmake/add_matlab_unit_tests.pl /tmp/drake-5642/drake
Can't open perl script "/tmp/drake-5642/drake/cmake/add_matlab_unit_tests.pl": No such file or directory

billhoffman, 2016-03-25:
@RussTedrake can you untar this in /tmp/t
[t.tar.gz](https://github.com/RobotLocomotion/drake/files/189972/t.tar.gz)
Then:
cd /tmp/t
mkdir b
cd b
cmake ..
ctest -VV

I get this:
Run command: /usr/bin/perl /tmp/t/add_matlab_unit_tests.pl /tmp/t
/tmp/t

hello
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: test

1: Test command: /usr/bin/cmake "-E" "echo" "hello"
1: Test timeout computed to be: 1500
1: hello
1/1 Test #1: test .............................   Passed    0.01 sec

RussTedrake, 2016-03-26:
i get 
```
russt@drake006:/tmp/t/b$ ctest -VV
UpdateCTestConfiguration  from :/tmp/t/b/DartConfiguration.tcl
Parse Config file:/tmp/t/b/DartConfiguration.tcl
 Add coverage exclude regular expressions.
UpdateCTestConfiguration  from :/tmp/t/b/DartConfiguration.tcl
Parse Config file:/tmp/t/b/DartConfiguration.tcl
Test project /tmp/t/b
Run command: /usr/bin/perl\ /tmp/t/add_matlab_unit_tests.pl\ /tmp/t
No such file or directory
Problem running command: /usr/bin/perl /tmp/t/add_matlab_unit_tests.pl /tmp/t
Problem executing pre-test command(s).
Errors while running CTest
```

RussTedrake, 2016-03-26:
drake006 is our best guard against breaking matlab still.  really need to resolve this one asap.

RussTedrake, 2016-03-26:
in case it helps:
```
russt@drake006:/tmp/t/b$ cmake --version
cmake version 2.8.11.2
```
and we're on ubuntu 12.04

RussTedrake, 2016-03-26:
and `b/CTestCustom.cmake` looks like this
```
set(CTEST_CUSTOM_PRE_TEST "/usr/bin/perl /tmp/t/add_matlab_unit_tests.pl /tmp/t")
```

RussTedrake, 2016-03-26:
also just verified that the newest version of cmake does not have this problem:
```
russt@drake006:/tmp/t/b$ /tmp/cmake-3.5.1-Linux-x86_64/bin/cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/t/b
russt@drake006:/tmp/t/b$ /tmp/cmake-3.5.1-Linux-x86_64/bin/ctest -VV
UpdateCTestConfiguration  from :/tmp/t/b/DartConfiguration.tcl
Parse Config file:/tmp/t/b/DartConfiguration.tcl
 Add coverage exclude regular expressions.
SetCTestConfiguration:CMakeCommand:/tmp/cmake-3.5.1-Linux-x86_64/bin/cmake
UpdateCTestConfiguration  from :/tmp/t/b/DartConfiguration.tcl
Parse Config file:/tmp/t/b/DartConfiguration.tcl
Test project /tmp/t/b
Run command: /usr/bin/perl /tmp/t/add_matlab_unit_tests.pl /tmp/t
/tmp/t

hello
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: test

1: Test command: /tmp/cmake-3.5.1-Linux-x86_64/bin/cmake "-E" "echo" "hello"
1: Test timeout computed to be: 1500
1: hello
1/1 Test #1: test .............................   Passed    0.00 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.00 sec
```

RussTedrake, 2016-03-26:
i've blasted my ppa install of cmake on drake006 to use 3.5.1 just to get the build server back up.  but i still think we need to resolve this.

billhoffman, 2016-03-26:
OK, I reproduced this with cmake 2.8.11.2 on my linux machine.  I was actually running 2.8.12.  Looks like a bug in 2.8.11 that affects ctest parsing that string that was broken and fixed in 2.8.12.

RussTedrake, 2016-03-26:
Great.  Then I think we just need to update the ppa for the ubuntu 12.04 installation instructions.

billhoffman, 2016-03-26:
Is there a 2.8.12 ppa?

RussTedrake, 2016-03-27:
You guys are the experts.  Also will want to increment the min version in the cmakelists

jamiesnape, 2016-03-28:
@billhoffman We would have to make one.

jamiesnape, 2016-03-28:
A PPA of 2.8.10 would probably be easier as that is the default on 14.04.

billhoffman, 2016-03-28:
2.8.10 seems to have the bug as well.  The binary from here for 2.8.12 works: https://cmake.org/files/v2.8/

closed 2016-04-05 by billhoffman