RobotLocomotion/drake #1905: Runtime error on Mac OSX El Capitan 10.11

erik-sobel, 2016-03-21:
The build seems to be using RPATH and can't find dynamic libraries at runtime.

 ./drake-visualizer
dyld: Library not loaded: @rpath/./libQtOpenGL.4.dylib
  Referenced from: ./build/bin/drake-visualizer
  Reason: image not found
Trace/BPT trap: 5

%otool -L drake-visualizer
All the libs are full path except these...
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtScript.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)

patmarion, 2016-03-21:
Is this something that you downloaded or compiled from source?  If downloaded, is it drake-v0.9.11-mac.tar.gz from the drake releases page?

erik-sobel, 2016-03-21:
Built from source from the latest distro.

patmarion, 2016-03-22:
Can you post the results of these two commands:

    $ which qmake
    $ qmake -v

The Director build is tested using Qt 4.8 installed from Homebrew.  Is it possible you have installed Qt from someplace else?  If you have homebrew qt installed, does the rpath string show up in the Qt libs, for example if you run:

    $ otool -L /usr/local/Frameworks/QtCore.framework/QtCore

erik-sobel, 2016-03-22:
Here's the result.

$ which qmake
/Users/esobel/anaconda/bin/qmake
$ qmake -v
QMake version 2.01a
Using Qt version 4.8.6 in /Users/esobel/anaconda/lib

$ otool -L /usr/local/Frameworks/QtCore.framework/QtCore
/usr/local/Frameworks/QtCore.framework/QtCore:
	/usr/local/opt/qt/lib/QtCore.framework/Versions/4/QtCore (compatibility version 4.8.0, current version 4.8.7)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 48.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1256.14.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 57337.20.44)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 728.6.0)

I suspect that anaconda could be the problem since I had to remove my anaconda3 package (python3) to get drake to even build.

patmarion, 2016-03-22:
Ok, try to prepend /usr/local/bin to your PATH so that "which qmake" shows the homebrew version under  /usr/local.  Then cd to drake/externals/director and run:

    make clean
    make drake-distro 

That will do a clean rebuild of the director external, and it should use homebrew Qt now.

erik-sobel, 2016-03-22:
Awesome!  it worked.  Thanks!

closed 2016-03-22 by erik-sobel