RobotLocomotion/drake #1906: Compilation issues

aespielberg, 2016-03-22:
Hi all,

I'm trying to compile Drake on Win10 64 bit.  I worked with Twan to get this installed on another PC, and basically replicated the steps on another PC, but I'm getting a few errors in the compilation.  In short, my project is set up as such:

[image]

cmake runs with no errors.  I then run make, and I get a few errors.  For example, near the end:

```
Build FAILED.

"C:\ResearchCode\drake\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\ResearchCode\drake\pod-build\bot_core_lcmtypes.vcxproj" (default target) (3) ->
"C:\ResearchCode\drake\pod-build\lcm.vcxproj" (default target) (5) ->
(CustomBuild target) ->
  ..\lcm/windows/WinPorting.h(29): error C2059: syntax error: 'string' [C:\ResearchCode\drake\externals\lcm\lcm\lcm-src
\pod-build\lcm-python\install_python.vcxproj] [C:\ResearchCode\drake\pod-build\lcm.vcxproj]
  CUSTOMBUILD : error : command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\amd64\\cl.exe' failed
with exit status 2 [C:\ResearchCode\drake\externals\lcm\lcm\lcm-src\pod-build\lcm-python\install_python.vcxproj] [C:\Re
searchCode\drake\pod-build\lcm.vcxproj]
  ..\lcm/windows/WinPorting.h(29): error C2059: syntax error: 'string' [C:\ResearchCode\drake\externals\lcm\lcm\lcm-src
\pod-build\lcm-python\install_python.vcxproj] [C:\ResearchCode\drake\pod-build\lcm.vcxproj]
  CUSTOMBUILD : error : command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\amd64\\cl.exe' failed
with exit status 2 [C:\ResearchCode\drake\externals\lcm\lcm\lcm-src\pod-build\lcm-python\install_python.vcxproj] [C:\Re
searchCode\drake\pod-build\lcm.vcxproj]

    0 Warning(s)
    4 Error(s)

```
I'm not sure if these are issues with Windows 10 or if  I have misconfigured something.  Do people have any suggestions of how to debug this?  It seems like all the bugs are with lcm.

-Andy S.

jhoare, 2016-03-23:
Its possible you have the same issue as encountered in #1900.

When building 64 bit are you setting CMAKE_FLAGS to -G "Visual Studio 14 2015 Win64" like in: http://drake002.csail.mit.edu/drake/sphinx/windows.html ?

billhoffman, 2016-03-23:
It is using the 64 bit compiler: BIN\\amd64\\cl.exe, the error here is caused by C code seeing extern "C".  This is python trying to compile C code  to create the python egg.   It most likely is an issue with the python you have installed.

billhoffman, 2016-03-23:
Here is a 32 bit build on my machine:
2>------ Build started: Project: install_python, Configuration: Debug Win32 ------
2>  Building Custom Rule C:/Users/hoffman/Work/drake/draketest/externals/lcm/lcm/lcm-src/lcm-python/CMakeLists.txt
2>  CMake does not need to re-run because C:\Users\hoffman\Work\drake\draketest\externals\lcm\lcm\lcm-src\pod-build\lcm-python\CMakeFiles\generate.stamp is up-to-date.
2>  running build
2>  running build_py
2>  running build_ext
2>  running install
2>  running install_lib
2>  running install_egg_info
2>  Removing C:\Users\hoffman\Work\drake\draketest\build\Lib\site-packages\lcm-1.3.0-py2.7.egg-info
2>  Writing C:\Users\hoffman\Work\drake\draketest\build\Lib\site-packages\lcm-1.3.0-py2.7.egg-info

jhoare, 2016-03-23:
Now that #1903 is merged, can you update and see if the error still occurs?

aespielberg, 2016-03-24:
This got further, but there is still an error.  

```
CUSTOMBUILD : CMake error : The source "C:/ResearchCode/drake/drake/CMakeLists.txt" does not match the source "C:/Resea
rchCode/drake/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory. [C:\ResearchCode
\drake\pod-build\drake.vcxproj]
```

and then later...

```
"C:\ResearchCode\drake\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\ResearchCode\drake\pod-build\drake.vcxproj" (default target) (19) ->
(CustomBuild target) ->
  CUSTOMBUILD : CMake error : The source "C:/ResearchCode/drake/drake/CMakeLists.txt" does not match the source "C:/Res
earchCode/drake/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory. [C:\ResearchCo
de\drake\pod-build\drake.vcxproj]

    0 Warning(s)
    1 Error(s)
```

I'm not sue why this error would appear.  This is from a fresh pull from a fork that I have updated.  I can't imagine that pull-request merge is messed up at all.  I'm wondering if I have my source/build folders set incorrectly?  My source is set to:

/drake
and my build is set to:
/drake/drake/pod-build

Is there something else I'm doing wrong?

aespielberg, 2016-03-28:
@jhoare @billhoffman , are you guys in Russ's group or are you outside collaborators?  I want to try to get this compiled ASAP, and if it's easier, I can swing by in person.  I can work on debugging these build errors but I need some intuition as to what the failure mode is here.  I'm not understanding really why there would be conflicting CMakeLists here.

jhoare, 2016-03-28:
Sorry we aren't located at MIT. I'm not sure what's going on in your new error, but I would try to do a complete clean and start over. 

Alternatively, you could delete all the ``build`` and ``pod-build`` folders that exist within drake, and start again.

aespielberg, 2016-03-28:
I have done several complete new clones from scratch :-).  Let me try it not from my fork, just as a sanity check.  Do you note have these problems on your side from a fresh clone?

jhoare, 2016-03-28:
No things are running on our end. I would guess that the issue is you may have some library installed different from our setup that CMake is finding and is causing issues.

billhoffman, 2016-03-28:
This is what is wrong:
/drake
and my build is set to:
/drake/drake/pod-build

Should be:
/drake
and my build is set to:
/drake/pod-build

That will create a /drake/drake/pod-build, but you need to run cmake in drake/pod-build for things to work.

I would say, complete clean and git clone.  Then build, you will have to have pkg-config in your PATH as well as cmake and make for things to work.  If you use anything other than the default cmake generator you will need to add -G into the CMAKE_FLAGS env var.

aespielberg, 2016-03-29:
hi @billhoffman, Thank you, it wound up being that I had the CMAKE_FLAGS environment variable incorrectly.  Thanks for putting up with the dumb questions, and sorry!

closed 2016-03-29 by aespielberg