RobotLocomotion/drake #1908: Modify Car Sim's Visualizer to Also Show Collision Model and Contact Reaction Forces

liangfok, 2016-03-22:
Having this would help debugging! In Gazebo, visualizations of both can be toggled on/off:

[image]

In the screenshot above, the brown cylinders visualize the collision models and the green vectors show the contact forces.

patmarion, 2016-03-22:
drake-visualizer can be made to draw things like this if the BotVisualizer class in drake publishes the information over lcm.  Currently, I think only geometry belonging to a RigidBodyTree is published, but not auxilary data like coordinate frames, contact points, collision models, force vectors, etc.

The drake sim could publish this information as drawing commands, or it could the information could be published as general simulation data, so that processes other than a visualizers could subscribe and consume the data.  Not sure if that's a real use case, but just something to think about when designing messages.

liangfok, 2016-03-22:
Perhaps publishing the contact / collision information as dedicated messages would be useful for planners to know when to, for example, transition between different contact configurations (assuming planners are running as a separate process from the controller / simulator)?

zamzami, 2016-03-22:
This is a very useful feature in practice for legged systems as well as other vehicles. So in addition to what @liangfok has mentioned, an example of "a real use case" @patmarion could be vehicle handling control. So imagine controlling roll stability while overtaking a corner at high speed or driving over rough terrain. It is useful to visualize, for example, the controller effect on the Lateral Load Transfer (LLT). 

check out the demo of MSC Adams: 
https://youtu.be/ypKkeX2IRNY?t=937
https://youtu.be/ypKkeX2IRNY?t=277

There is already a hacky version in matlab `classdef ContactWrenchVisualizer < BotVisualizer` in`ContactWrenchVisualizer.m` but it is hardcoded to Atlas

amcastro-tri, 2016-03-25:
I was talking to @RussTedrake today and he pointed me to a very neat library already available with Drake! It is called lcmgl and it is located [here](https://github.com/RobotLocomotion/libbot/tree/master/bot2-lcmgl). You can find it under your drake-distro in externals/libbot/bot2-lcmgl/src/bot_lcmgl_client/.

A good place where to start, pretty well documented in the source, is lcmgl.h. This essentially allows you to send open gl commands over lcm that director can interpret. We can even make this calls anywhere in the code just for debugging! 

I'll give it a try maybe tomorrow to debug my suit of test cases for #1616.

jwnimmer-tri, 2017-06-19:
Re-assigning to be consistent with "team : robot locomotion group".

RussTedrake, 2017-06-19:
i think this should remain assigned to @liangfok or be closed.  it is not a request from the locomotion group.

jwnimmer-tri, 2018-06-09:
https://github.com/RobotLocomotion/drake/blob/master/multibody/rigid_body_plant/visualization/contact_viz.py

closed 2018-06-09 by jwnimmer-tri