RobotLocomotion/drake #1910: AWS builds all have this error cannot stat ‘ccache-swig.1

billhoffman, 2016-03-22:
/usr/bin/install: cannot stat ‘/home/ubuntu/workspace/linux-clang-continuous/externals/swig_matlab/swig-matlab/CCache/ccache-swig.1’: No such file or directory
make[8]: *** [install] Error 1
make[7]: *** [install-ccache] Error 2

https://drake-jenkins.csail.mit.edu/job/experimental/358/compiler=clang,label=linux/consoleFullhttps://drake-jenkins.csail.mit.edu/job/experimental/358/compiler=clang,label=linux/consoleFull

rdeits, 2016-03-22:
The actual issue is the lack of yodl2man on those build servers, as I mentioned here: https://github.com/RobotLocomotion/drake/pull/1825#issuecomment-193951619

jamiesnape, 2016-03-22:
Since it is not a prerequisite, it is not installed.

rdeits, 2016-03-22:
Yes, and it's not actually necessary for the build. The error message is just unfortunate noise.

billhoffman, 2016-03-22:
Right now the jenkins scripts miss this type of error.   I would like to fix that, but then all builds would be broken for PR's on linux.  So, I want to come up with a fix for this so I can parse for this type of error.

jamiesnape, 2016-03-22:
I can install `yodl` on Ubuntu, but we will get the issue on other platforms.

billhoffman, 2016-03-22:
I just logged into one of the build servers, seems that just one file is missing....  If I do this:
touch /home/ubuntu/workspace/experimental/compiler/gcc/label/linux/externals/swig_matlab/swig-matlab/CCache/ccache-swig.1

Then it works.  We might be able to just create that file in the right place with the external project.

billhoffman, 2016-03-22:
No, that does not work...  You still get this error:make[6]: yodl2man: Command not found

jamiesnape, 2016-03-22:
There was this patch on a fork of the original swig: https://github.com/ahnolds/swig/commit/cd0b4b5d793c6926c4f35fa8f42285bc23b2b2e6

jwnimmer-tri, 2016-03-22:
(Sorry if this isn't the right ticket for this discussion, please point me somewhere else if needed.)

> Right now the jenkins scripts miss this type of error.

How is this even possible?  Shouldn't any and all errors seen by make turn into a non-zero exit status that will be propagated back up to the topmost level, which jenkins would see and declare a build failure?  I don't understand why (apparently) text scraping the build logs is ever a good idea.

jamiesnape, 2016-03-22:
The build command for that project has `|| true` appended to it.

jamiesnape, 2016-03-22:
Scraping build logs is how CTest finds warnings.

billhoffman, 2016-03-22:
It is also because we are running all the builds via ctest which sends the output to the cdash dashboard ctest almost always returns non-zero even if one test fails.  So, for Jenkins we have to text scrape to get the errors.  Actually, on linux with ctest-launchers enabled, CTest examines the return value of each command run.   It is the mixture of Jenkins and CTest that requires us to look at the text.  That said if the || True was not there, it would have been a build error and we would have caught it.  

All that said, sometimes we have to look at the whole log file from Jenkins and I always search for "error " and I always find this stuff, so it is a distraction.  And it is possible that we could miss some other type of error in the future if we do not explicitly look for this type of error.

billhoffman, 2016-03-22:
Right now, the external project uses this fork of swig: https://github.com/jaeandersson/swig.  Maybe we could change it to use one that has the patch found here:  ahnolds/swig@cd0b4b5.   Or we could provide a patch and use git apply to put it in place.

rdeits, 2016-03-22:
FYI @billhoffman we're using that fork of swig because it is the only one that supports generating MATLAB bindings

billhoffman, 2016-03-22:
@rdeits who maintains that fork?  Would they be open to merging in the patch that @jamiesnape found that disables this man page stuff?

jamiesnape, 2016-03-22:
Apparently this was also applied to the main swig repo: https://github.com/swig/swig/commit/64dcd50b9953cad388db44f772c7dbee1cb34b7d

billhoffman, 2016-03-22:
Cool, we just need that merged into  https://github.com/jaeandersson/swig

jamiesnape, 2016-03-22:
And it was applied here: https://github.com/jaeandersson/swig/commit/64dcd50b9953cad388db44f772c7dbee1cb34b7d

@rdeits just needs to update the hash in the pod.

jamiesnape, 2016-03-22:
And in drake.

jamiesnape, 2016-03-22:
And remove the `|| true`.

rdeits, 2016-03-22:
See #1913

RussTedrake, 2016-03-23:
@jamiesnape -- this will be closed by #1913, but the || true has not been removed.

jamiesnape, 2016-03-23:
Looks like it was removed here:

https://github.com/rdeits/swig-matlab-pod/commit/86f2c16d9b27489ebf28982c6d3a98ea0de07ce5

RussTedrake, 2016-03-23:
Ah, great.

PR #1913 (2016-03-23): update swig version to fix yodl issue
fixes #1910
commit d1335319 (2016-03-23): update swig version to fix yodl issue

fixes #1910
closed 2016-03-23 by RussTedrake