RobotLocomotion/drake #1914: Parameter parsing in cpp code/RigidBodyTree tests

aespielberg, 2016-03-22:
Is there currently any parameter parsing in the urdf parsing in the cpp side of things?

The only place I see parameters in the C++ code is in constructModelMex - I see a vector of doubles called params_vec.  I'm fairly certain this isn't quite the same.  So, a few other quick questions on this note:

-There are a lot of rigidbody and rigidbodytree classes.  Is there any documentation on the purpose of these?  
-Is there a RigidBodyTree test anywhere that I can look at to step through a parsing example?
-Where is the entry point to the urdf parsing?

aespielberg, 2016-03-22:
Oh, I did find udfKinTest, so that answers the last two questions.

liangfok, 2016-03-22:
> Is there currently any parameter parsing in the urdf parsing in the cpp side of things?

I believe most of the parameters are parsed by methods in [xmlUtil.h](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/xmlUtil.h).

liangfok, 2016-03-22:
> -There are a lot of rigidbody and rigidbodytree classes. Is there any documentation on the purpose of these?

Which classes are you referring to? I'm only aware of two: [RigidBody.h](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBody.h) and [RigidBodyTree.h](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTree.h).

aespielberg, 2016-03-22:
Ah, I see.  So there's jut RigidBodyTree and RigidBody, and then RigidBodyTreeContact, RigidbodyTreeURDF, and RigidBodySDF are all just implementations of functions and methods, it seems, looking closer.  Thanks!

aespielberg, 2016-04-02:
I am reopening this, as upon inspection there does not seem to be parameter parsing on the CPP side.  I will work on implementing this.

RussTedrake, 2016-09-08:
resolving as duplicate of #2030

closed 2016-09-08 by RussTedrake