RobotLocomotion/drake #1916: Make valuecheckMatrix googletest-friendly

david-german-tri, 2016-03-22:
[valuecheckMatrix](https://github.com/RobotLocomotion/drake/blob/master/drake/util/testUtil.h#L80) contains useful logic for comparing two Eigen matrices.  This should be migrated into a googlemock matcher if possible, or at least modified to return a boolean rather than assert.

liangfok, 2016-03-22:
Relevant discussion here: https://github.com/RobotLocomotion/drake/issues/1887#issuecomment-198778371. I'm not familiar with a "googlemock matcher". Will look into it.

liangfok, 2016-03-24:
I think I need this to address some of the code review comments in https://github.com/RobotLocomotion/drake/pull/1898, so I assigned it to myself.

liangfok, 2016-03-24:
I decided to convert one unit test that uses `valuecheckMatrix()` to use the Google Test framework, and then update it to use a Google Mock. [Here is the current state.](https://github.com/RobotLocomotion/drake/compare/master...liangfok:feature/matrixMatcher?expand=1)

The current problem is reported during compile time:

```
/Users/liang/dev/drake/build/include/gmock/gmock-matchers.h:643:5: error: static_assert failed "T_must_be_implicitly_convertible_to_U"
    GTEST_COMPILE_ASSERT_((internal::ImplicitlyConvertible<T, U>::value),
    ^                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
The above error is due to me using the mock to compare an `Eigen::Vector2d` to an `Eigen::VectorXd`. I believe both of these classes derive from `Eigen::MatrixBase`` so thought it should work. Will continue to look into this problem.

david-german-tri, 2016-03-24:
Possibly you want a [polymorphic matcher](https://github.com/google/googletest/blob/master/googlemock/docs/CookBook.md#writing-new-polymorphic-matchers)?

liangfok, 2016-03-25:
Thanks! The [polymorphic matcher worked](https://github.com/google/googletest/blob/master/googlemock/docs/CookBook.md#writing-new-polymorphic-matchers). I will complete the matcher's implementation, replace the old `valuecheckMatrix()` with the new matcher and then submit a pull request.

**Question:** Can Google Mock matchers be used in "core" Drake software like `RigidBodyTree` and `RigidBody` or are they restricted to unit tests? I need to access the matrix-equality-check logic in these methods' `operator==` methods.

david-german-tri, 2016-03-25:
Nice!  googlemock should only be used in tests, but you could factor out the equality logic into a function that's called from both your matcher and from production code.

liangfok, 2016-03-26:
I finished updating all of the unit tests to use `CompareMatrices(...)` instead of `valueCheckMatrix(...)`. Still need to run cpplint and clang-format on the code I touched before creating a PR.

Having trouble updating [Reviewboard](https://rbcommons.com/s/tri/r/2/). Not sure why but `rpt` disappeared from my computer.

closed 2016-03-31 by david-german-tri