RobotLocomotion/drake #1919: CI Servers Should Save Verbose Output

liangfok, 2016-03-23:
Can the CI servers use "make VERBOSE=true" so we know exactly what compiler / linker commands are being executed?

jamiesnape, 2016-03-23:
It is enabled for new builds on the server. The flag may or may not get propagated to the externals via pods. We will see shortly.

billhoffman, 2016-03-23:
CDash already saves that information at least for makefile builds on unix because it uses ctest launchers which capture each command.  For example, click the + on the command line here:
https://drake-cdash.csail.mit.edu/viewBuildError.php?type=1&buildid=3030
Shows the full command line for the compile.

jamiesnape, 2016-03-23:
@liangfok Is the CDash support sufficient or would you prefer builds to be verbose in the Jenkins console too?

jwnimmer-tri, 2016-03-23:
As a specific example, PR #1898 failed, at [link](https://travis-ci.org/RobotLocomotion/drake/builds/118062191).  How to we view the verbose output for that build failure?

liangfok, 2016-03-23:
@jamiesnape: ditto what @jwnimmer-tri said. More verbose output in Travis CI's test would be helpful.

jamiesnape, 2016-03-23:
Ok. I will have to put in a PR to make Travis CI more verbose.

PR #1922 (2016-03-26): Make CI builds more verbose
Closes #1919.
closed 2016-03-26 by RussTedrake