RobotLocomotion/drake #1920: runQuadrotorDynamics Unit Test Fails When Drake is Compiled With Debug Symbols

liangfok, 2016-03-23:
Here is the error:

```
2/39 Test #32: runQuadrotorDynamics ..............................***Exception: Other  3.00 sec
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Simulation is not keeping up with desired real-time factor -- behind by more than 1 (scaled) second at simulation time 0.140000
```

I believe it's because the inclusion of debug symbols results in the unit test running too slowly. Either the unit test should be disabled or the time limit should be increased when Drake is compiled with debug symbols.

RussTedrake, 2016-03-23:
agreed. this has been failing. would appreciate it if you can resolve 
(increasing the time limit seems ok to me)

psiorx, 2016-03-23:
I don't think increasing the time limit will fix this.  The unit test isn't timing out.  It's outright failing because it's configured to keep up with real-time which it can't do in Debug mode.

RussTedrake, 2016-03-23:
of course you’re right. good call. we should pass in simulation options so it does not try to run at real-time.

jwnimmer-tri, 2016-03-26:
Even in release builds, this test appears to randomly fail in CI.

jwnimmer-tri, 2016-03-26:
@RussTedrake This is just as simple as changing examples/Quadrotor/runDynamics.cpp:81 to `options.realtime_factor = 0.0;` hard-coded, instead of 1.0?  Or did you mean you want the build system to use argv to override that, but still leave the default at 1.0?

RussTedrake, 2016-03-26:
Command line would be the better resolution.
One other option: we could add one additional flag to the simulation logic 
which turns the error into a warning (would need a warnOnce-type logic) if 
we fall behind the desired speed.

liangfok, 2016-04-01:
> Even in release builds, this test appears to randomly fail in CI.

I have also noticed this on occasion and attribute it to the fact that we are not using real-time-capable operating systems. Here are some options:

1. **fair / easy:** Relax the latency threshold at which the unit test fails.
2. **better / hard:** Use a real-time-patched operating system and modify Drake to run at the highest priority level.
3. **best / insane:** Run Drake as the highest priority process in a real-time-patched operating system on a cycle-accurate simulator that allows us to deeply understand the computational latency of Drake in this context.

closed 2016-04-04 by RussTedrake