RobotLocomotion/drake #1927: RigidBodySystem shouldn't contain every specific RigidBodySensor

david-german-tri, 2016-03-24:
RigidBodySystem (.h and .cpp) has become pretty bloated.  @psiorx, would you consider moving your various RigidBodySensor subclasses to their own separate files?  It would also be a good opportunity to make them [style guide compliant](https://google.github.io/styleguide/cppguide.html).

@rpoyner-tri FYI

psiorx, 2016-03-24:
I agree.  I'm a strong believer of the one class per file rule.  Sensors should probably even have their own directory.

@RussTedrake Was there a specific reason for having multiple classes in the same file?  I've seen this in a few other places as well (CascadeSystem comes to mind).

RussTedrake, 2016-03-25:
Not a good reason (was just trying to push out a boat load of functionality 
before the influx of developers). Much better to have separate files.

david-german-tri, 2017-01-21:
This is moot now that System1 is gone.

closed 2017-01-21 by david-german-tri