RobotLocomotion/drake #1932: drake-visualizer Uses Deprecated Method userSpaceScaleFactor

liangfok, 2016-03-24:
I'm getting the following warning when I launch `drake-visualizer` on my OS X machine:

```
$ ./build/bin/drake-visualizer
2016-03-24 11:57:09.636 drake-visualizer[7779:1073592] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead.
```

liangfok, 2016-03-24:
The warning is due to OS X changing its API:

https://developer.apple.com/library/mac/documentation/Cocoa/Reference/ApplicationKit/Classes/NSWindow_Class/#//apple_ref/occ/instm/NSWindow/userSpaceScaleFactor

The `userSpaceScaleFactor` method was introduced in OS X 10.4 but deprecated in OS X 10.7.

Not sure what component within drake-visualizer uses this method. The same warning shows up when running other applications in OS X like [meld](http://meldmerge.org).

patmarion, 2016-03-26:
This is a warning coming from the Qt library.  It has been fixed in newer versions of Qt:

https://bugreports.qt.io/browse/QTBUG-28574

I'm going to close this issue because I don't think it is affecting anything.  By the way, you can report drake-visualizer bug reports here:

https://github.com/RobotLocomotion/director/issues

closed 2016-03-26 by patmarion