RobotLocomotion/drake #1933: RigidBodySystem contact force check scales poorly with scene size

david-german-tri, 2016-03-24:
The "contact forces" step in RigidBodySystem::dynamics does a collision check that is O(N^2) in the number of collision volumes in the RigidBodyTree.  In large static scenes with comparatively small robots, collision checks between pieces of terrain that never actually move or collide dominate the runtime.  In one example I'm tinkering with locally, the scene has 147 collision volumes and the robot has 4, so only about 2.6% of the collision checks Drake performs are actually useful.

We should have an interface to express "don't check for collisions between bodies in group B".  This interface might already exist - there are a lot of different collision detection calls in Drake at various layers of abstraction, and I haven't sorted through them all yet.

sherm1, 2016-03-24:
The Bullet collision system should handle this well, as will the one we're going to develop (tentatively based in FCL). No collision checks should be done among objects attached to the same rigid body, and there is no need for O(n^2) performance even when checking among objects that could potentially collide.

david-german-tri, 2016-03-24:
That's good to know.  The Drake wrappers around bullet impose O(N^2) performance [here](https://github.com/RobotLocomotion/drake/blob/5cbb4b5d61538613724d152cf75a0f7c71bea951/drake/systems/plants/collision/BulletModel.cpp#L638). I'll start digging into how to use the actual Bullet API to do better.

sherm1, 2016-03-24:
> The Drake wrappers around bullet impose O(N^2) performance [here](https://github.com/RobotLocomotion/drake/blob/5cbb4b5d61538613724d152cf75a0f7c71bea951/drake/systems/plants/collision/BulletModel.cpp#L638)

Yikes!

david-german-tri, 2016-03-24:
OK.  The problem I described above is exactly what Bullet masks were designed to solve - page 17 of the [Bullet 2.83 manual](https://github.com/bulletphysics/bullet3/blob/master/docs/Bullet_User_Manual.pdf).  We call `addCollisionObject` [here](https://github.com/RobotLocomotion/drake/blob/5cbb4b5d61538613724d152cf75a0f7c71bea951/drake/systems/plants/collision/BulletModel.cpp#L218) in a pretty straight line from parsing an SDF `<collision>` tag.

So, it seems like we can just add masks to `RigidBody::CollisionElement` and have a nice API on the Drake-to-Bullet side.  On the SDF-to-Drake side, it's not immediately clear what SDF fields should mean "hey, this stuff is all terrain, don't collide it with itself".  Stay tuned.

david-german-tri, 2016-03-24:
The Drake-to-Bullet masking is not as straightforward as I thought!  `RigidBodyTree::collisionDetect` calls `BulletModel::closestPointsAllToAll`, which runs an O(N^2) loop over the bodies in order to [call GJK on them directly](https://github.com/RobotLocomotion/drake/blob/5cbb4b5d61538613724d152cf75a0f7c71bea951/drake/systems/plants/collision/BulletModel.cpp#L468).  The entire Bullet collision world is essentially just used as a hash-map to store the btConvexShapes.  We're calling into Bullet at a low level where we can't use its masking, broadphase, or anything else.

@RussTedrake, is there something I should read, or someone I should talk to, to understand why it works this way?

RussTedrake, 2016-03-25:
Had this conversation today with @amcastro-tri, too.
The collision detection code was all built initially for motion planning and control, not for simulation. In that context, you want a fixed number of constraints, most commonly all n^2 — for instance so the motion planner can evaluate the signed distance between each pair of obstacles, and evaluate the impact on those distances given a change in the motion parameters.
The fact that we also started using this for simulation is purely historical. It is not the right way to do simulation, but it worked fine for simple problems. We will want to preserve that old interface, but also call bullet’s GJK directly.

amcastro-tri, 2016-03-25:
Yes, we talked about this with @RussTedrake today. @david-german-tri, I am actually in the process of adding the capability to use Bullet's collision dispatcher since the current implementation does not allow to have concave shapes (just as height maps) and also fails in some very simple tests I've been running even for convex shapes, see #1616. 

Bullet's filters is the mechanism you want to use in your case where a number of fixed objects forming part of the world are checked against each other when unnecessary. 

This will allow us to hopefully model contact between the car's wheels and the ground more accurately. I do agree however with @RussTedrake in that we need to keep the O(n^2) infrastructure since this is needed for the design of control algorithms.

amcastro-tri, 2016-03-25:
I'm in the process of creating a pull request for #1616 that will happen tomorrow morning (sorry, done for the day). With that I'll also upload a suit of test cases I created to test the accuracy of the collision and contact algorithm. Hopefully that will clarify things

david-german-tri, 2016-03-27:
As @amcastro-tri mentioned on #1940, I'm looking into using RigidBody::collidesWith to mask out terrain-on-terrain collisions.  SDF provides a [static](http://sdformat.org/spec?ver=1.6&elem=model#model_static) tag in the model element to indicate that a model is immovable.  So, I plan to add all static bodies to a group that does not self-collide.

RussTedrake, 2016-03-27:
I would have thought that objects marked static should get added directly 
to the world link (link 0), and that those would automatically get grouped 
into a single collision group. no?

avalenzu, 2016-03-27:
Also, the collidesWith method implements the same type of  masking that
Bullet uses. Adding rigid bodies A, B, and C to a collision filter group
"foo" and setting group "foo" to ignore collisions with itself will prevent
any collision checks from being performed between the collision elements of
A, B, and C. At one point the masks from Drake collision filter groups were
also written to the masks of the btCollisionObjects, but I'm not sure if
that's still the case. Also, I'm not sure if the Drake-specific
collision_filter_group tag is being parsed in the C++ URDF parser.

As the author of that O(n^2) dispatching, I just wanted to reiterate what
@RussTedrake was saying: that collision interface was written for
trajectory optimization, where we needed functions that were as smooth as
possible. In that context, the efficiency gains from more effective
filtering of potential collisions would have been eradicated by the
discontinuities from changing numbers of collision points. It's not a good
system for simulation, it just happened to already be there when we wanted
to start treating collisions between convex shapes. I'd love to see a
separate interface for simulation!

david-german-tri, 2016-03-27:
> Adding rigid bodies A, B, and C to a collision filter group "foo" and setting group "foo" to ignore collisions with itself will prevent any collision checks from being performed between the collision elements of A, B, and C.

@avalenzu, I'm only proposing to do that for rigid bodies that are static, in which case the behavior you foresee seems like the correct one.

> At one point the masks from Drake collision filter groups were also written to the masks of the btCollisionObjects, but I'm not sure if that's still the case

It isn't.

david-german-tri, 2016-03-28:
@RussTedrake I'm not sure I follow.  My impression was that a link is a RigidBody, and a RigidBody can only have one collision geometry.  So I'm not sure how I'd put a whole scene's worth of objects into the world link.  Am I missing something?

RussTedrake, 2016-03-28:
You can have multiple collision geometries for a single rigid body.

david-german-tri, 2016-03-28:
OK, I see that now.  Yes, we could modify the SDF parser to read all the `<link>`s from a static `<model>` into the world RigidBody.  It seems a little odd to me to break the 1:1 relationship between SDF links and RigidBody objects, but I'm happy to do it that way if you think it's better.

RussTedrake, 2016-03-28:
Another option is to add it as a fixed joint to the parent and add the logic to say fixed joints preserve the collision groups.  Perhaps that is cleaner.

I think if a static object descends from a non-static object, then that should be a parse error.

liangfok, 2016-03-28:
+1 for adding fixed joints to build a collision group while preserving individual links.

A possible challenge is to support only loading a local window to the world.

liangfok, 2016-03-28:
Won't there be numerous combinations of links that are not marked static and yet have no chance for collision? For example, a "front bumper" link and  "rear bumper" link will in most normal circumstances not collide...

RussTedrake, 2016-03-28:
Yes.  We added support for collision filter groups from urdf.  Search the issues for a recent discussion about whether that data should exist in urdf or not

amcastro-tri, 2016-03-28:
It looks like SDF supports collision bitmasks (the recommended filter mechanism for static objects that do not collide under any circumstance). See  #1793. And [SDF specification for bitmasks](http://sdformat.org/spec?ver=1.6&elem=collision#contact_collide_bitmask)

sherm1, 2016-03-28:
> It looks like SDF supports collision bitmasks 

It is unfortunate that they put bitmasks in the model file! All that's needed from the modeler is to define some groups and assign membership in those groups; a bitmask would be one way to implement that but isn't the nicest way to specify grouping.

david-german-tri, 2016-04-17:
@amcastro-tri I think you're likely to fix this as part of your broader collision rethink, so I've reassigned it to you.  Feel free to kick it back to me if you disagree.

amcastro-tri, 2016-04-18:
@david-german-tri. I am working on very similar things right now that will hopefully help us in this regard also. I won't doubt kicking this back to you if I need to :-)

tri-ltyyu, 2018-11-09:
Per issue scrub, this is no longer relevant since this functionality is in SG. Closing

closed 2018-11-09 by tri-ltyyu