RobotLocomotion/drake #1937: "still reachable" events should not count as warnings on nightly Valgrind

david-german-tri, 2016-03-24:
Many of the errors being reported by in the Valgrind [nightly runs](https://drake-cdash.csail.mit.edu/viewDynamicAnalysis.php?buildid=3091) are about memory that is still reachable.  Can we suppress those?

jamiesnape, 2016-03-24:
We should probably move this file into the main repo and add to it:
https://github.com/billhoffman/drake/blob/build-scripts/scripts/valgrind.supp

jwnimmer-tri, 2016-03-24:
Yup.  A drake developer must be able to reproduce the CI results of valgrind on their workstation.  Ideally with a `make` target or option, including process-exit-status pass/fail outcome.

jamiesnape, 2016-03-24:
Remember that the CI infrastructure is a work in progress. Once it is stable and reliable many of the scripts that reside elsewhere will be added to this repo.

jwnimmer-tri, 2016-03-24:
Sounds good, I was missing that context.  Thanks!

jwnimmer-tri, 2016-03-25:
Though #1938 is a good step, for this issue I think the desire is to invoke valgrind with `--show-leak-kinds=definite,possible` instead of `all`.  According to the manual, that was already the default, so I'm not sure why we're getting "still reachable" errors reported in the first place.

It appear that valgrind.supp would be able to express "don't report Leak/match-leak-kinds:reachable for no matter what" so we could solve this with the valgrind.supp file.  However, that seems like an abuse of the tool.  I don't want to suppress reachable leaks; I just don't want to even look for them in the first place.

jamiesnape, 2016-03-25:
I will look into adding the command line flags.

jamiesnape, 2016-03-25:
Still reachable warnings are no longer showing. Two failing tests are being reported when running under Valgrind.

closed 2016-04-07 by jamiesnape